espruino/Espruino #473: SPI does not work on NucleoF401

SpenceKonde, 2015-01-14:
Tested with two SPI devices, the AT25 512 EEPROM and a PCD8455 LCD display. 

Using the example code from the module documentation, neither of these is willing to work on the Nucleo F401. 

Same code worked when the same device was connected to the Espruino Board (only changes were pin assignments). 

Software SPI didn't work either - neither of the devices would work with it. I am adding an issue about a dramatic failure of software SPI in another issue. 

Using 1/13 build from drazzy.com/e/espruino/NucleoF401

I suspect this issue will be relevant to the pico as well, as they both use an F401.

SpenceKonde, 2015-01-14:
The symptom was that the PCD8544 display didn't seem to do anything. The EEPROM appeared to return zeros (as will happen with nothing connected). I did try several different sets of pins - C10/11/12, B3/4/5, B13/14/15, as well as software SPI, which generated an assert failure.

gfwilliams, 2015-01-14:
Thanks - I guess this is just the same issue as: https://github.com/espruino/Espruino/issues/470

commit 1810f97d (2015-01-15): Fix I2C/SPI on F401/F411-based boards (fix #470, fix #473)
closed 2015-01-15 by gfwilliams