espruino/Espruino #474: Software SPI doesn't work on Nucleo F401, leads to ASSERT() FAILED

SpenceKonde, 2015-01-14:
Code:

```

var newSPI = new SPI();
newSPI.setup({ sck:C10, miso:C11, mosi:C12 });
var g = require("PCD8544").connect(newSPI,D2,B6,B7, function() {
  g.clear();
  g.drawString("Hello",0,0);
  g.drawLine(0,10,84,10);
  g.flip();
});

```

Using v72 1/13 nucleo build:

send the code and:

```

>reset();
=undefined
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v72 Copyright 2014 G.Williams
>echo(0);
=undefined
>g.flip();
=undefined
>g.flip();
ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l2] Object {
    #2[r1,l2] Name String [1 blocks] "ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l3] Object {
    #2[r1,l4] Name String [1 blocks] "ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l4] Object {
    #2[r1,l6] Name String [1 blocks] "ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l5] Object {
    #2[r1,l8] Name String [1 blocks] "ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l6] Object {
    #2[r1,l10] Name String [1 blocks] "ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l7] Object {
    #2[r1,l12] Name String [1 blocks] "ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l8] Object {
    #2[r1,l14] Name String [1 blocks] "ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l9] Object {
ASSERT() FAILED AT src/jsvar.c:416
  #1[r1,l10] Object { ASSERT() FAILED AT src/jsvar.c:402
  #1[r1,l11] Object { ASSERT() FAILED AT src/jsvar.c:402
...

```

Software SPI seems to still work on the normal Espruino board (both tests were conducted with my 1/13 wiznet builds)

gfwilliams, 2015-01-14:
Ohh, nasty. Thanks!

gfwilliams, 2015-01-14:
By the way, if you're getting asserts it means you're building without `RELEASE=1`. Code size will go down and speed will go up if you add it :)

gfwilliams, 2015-01-14:
This one turned out to be a complete bastard to fix - thanks for finding it though. It was actually part of a much bigger problem.

Actually I know I mentioned `RELEASE=1`, but if you could **not** use that for now it'd be amazingly helpful. I added a few more asserts, and it's possible that over the course of day to day usage people might hit those - which would really help with finding other bugs :)

closed 2015-01-14 by gfwilliams

SpenceKonde, 2015-01-14:
I was going to ask you about that. 

I didn't know that I had to explicitly say RELEASE=1 for nucleo builds (I think for bigram builds, they're always built release, since I use the create_espruino_image script to create), but considering that the F401/411 is a new board, I was inclined to leave debugging in for now anyway. 

I won't do RELEASE=1 until things settle down more on these boards.

Is there a way to do a debug (ie not RELEASE=1) build for the Espruino Board? I can make the bigram builds debug if you want (or build debugs in addition to normal ones). What would I need to change to do this?  Currently the commands I call for the bigram builds are:

(here, var/espbuild/ESPRUINOBOARD.py has the modified ram settings)

```

rm -rf /var/espbuild/espruino
git clone https://github.com/espruino/espruino /var/espbuild/espruino
cp /var/espbuild/ESPRUINOBOARD.py /var/espbuild/espruino/boards/ESPRUINOBOARD.py

/var/espbuild/espruino/scripts/create_espruino_image_1v3.sh

```

gfwilliams, 2015-01-15:
You should just be able to edit `/var/espbuild/espruino/scripts/create_espruino_image_1v3.sh` and remove the line `export RELEASE=1`. Having said that, the bootloader only thinks it has 256kB of flash, so it won't erase the pages at the end (as you found).

That means that if the full binary size (including the 16kB for bootloader) gets larger than 256kB then you won't be able to flash it using the USB bootloader and will have to use the serial one instead - which would probably annoy a bunch of people :)

commit 444ee48f (2015-01-14): Fix issues with freeing of variables not clearing the lock flags (fix #474)