espruino/Espruino #479: Make I2c return Uint8Arrays

gfwilliams, 2015-01-16:
it currently returns arrays, but Uint8Array would be a lot faster and more efficient.

http://forum.espruino.com/conversations/260971/#12045483

This may have sucked initially, but now Uint8Array supports basically all the same stuff as Array, so it should be much more sensible.

AlexanderBrevig, 2015-01-20:
I sat down and tried to make this change until I realized I don't have a way to test I2C right now, and I need to do other things. I'm still new so this code might be full of errors.

I post it here so I can pick it up again when I'm able to test - or provide a starting point for someone else in the meanwhile.

My intuition would be to change the code from from [jswrap_spi_i2c.c]
(https://github.com/espruino/Espruino/blob/master/src/jswrap_spi_i2c.c#L681):
```c
JsVar *array = jsvNewWithFlags(JSV_ARRAY);
if (array) {
  int i;
  for (i=0;i<nBytes;i++) {
    JsVar *v = jsvNewFromInteger(buf[i]);
    jsvArrayPushAndUnLock(array, v);
  }
}
```
To this Uint8Array code:
```c
JsVar *lenVar = jsvNewFromInteger(nBytes);
JsVar *array = jswrap_typedarray_constructor(ARRAYBUFFERVIEW_UINT8, lenVar,0,0);
jsvUnLock(lenVar);
if (array) {
  unsigned int i;
  for (i=0;i<(unsigned)nBytes;i++) {
    JsVar *v = jsvNewFromInteger(buf[i]);
    jsvArrayBufferSet(array, i, v);
    jsvUnLock(v);
  }
} //do we need to [un]lock array?
```

gfwilliams, 2015-01-20:
Thanks - that looks good to me. I'd suggest you use JsvArrayBufferIterator to set the contents though - it'll be faster, and there's a function specifically for setting integers, which doesn't require you to create the var with jsvNewFromInteger.

gfwilliams, 2015-01-20:
Just done this - thanks!

closed 2015-01-20 by gfwilliams

AlexanderBrevig, 2015-01-20:
Nice! Great refactors as well :) Happy to be of (possibly) some help

commit abc326d6 (2015-01-20): Added E.toString and E.toUint8Array
            Made I2C.readFrom return a Uint8Array (fix #479)