espruino/Espruino #481: Missing complement to digitalPulse

AlexanderBrevig, 2015-01-21:
A way to measure incoming pulses. 

Maybe something like:

```c
digitalPulseIn(pin:Pin, peak:bool, timeout: ms)
```

So - we'd tell the function:
```
- what pin to use
                    __                          __   __         
- to look for a   _/  \_ (true/high) pulse or a   \_/   (false/low)

- timeout (if no such pulse is detected, bail out - default should be??)
```

I think it makes sense if this is a blocking operation? I.e a timeout of 500ms will worst-case block ~500ms

gfwilliams, 2015-01-21:
I'm not convinced about this one I'm afraid - pretty much everything is designed to be nonblocking, as blocking for 500ms would usually cause a whole bunch of buffers to overrun.

Also consider this:

```
digitalPulse(A0, 1, 0.05);
var l = digitalPulseIn(A1, true, 500);
```

If you've got something connected that responds to A0 and pulses A1, there's absolutely no guarantee that *in the time taken to parse and execute the command*, which could be ~0.25ms, the pulse hasn't already happened.

It's really not too hard to do this with setWatch either:

```
var w = setWatch(A1, function(e) {
  var d = e.time - e.lastTime;
  // ...
}, { repeat: false, edge:"falling" });
```

AlexanderBrevig, 2015-01-21:
Just so you know I'm not stirring up trouble for no reason - I need this for using ultrasonic range sensors. My current plan for a work-around is to make a small AtTiny thingy that talks i2c / onewire and buffers in/out pulses TinyPulseer™ ;)

After struggeling a while to get it working through c I tried this:

```js
setTime(0);
var w = setWatch(function(e) {
  if (e.state === false) {
    console.log("pulse: " + (e.time - e.lastTime));
    console.log(e);
  }
}, BTN1, { repeat: true, edge:"both" });
```

Do you have a guesstimate for what kind of precision one might expect from that? 

Only problem is I get very weird values after a few cycles, am I doing something odd or what do you think?
```
pulse: 0.03183269500
{ "lastTime": 1.68449020385, "time": 1.71632289886, "pin": B12, "state": false }
pulse: 0.05562591552
{ "lastTime": 1.96714496612, "time": 2.02277088165, "pin": B12, "state": false }
pulse: 0.03679943084
{ "lastTime": 2.33329105377, "time": 2.37009048461, "pin": B12, "state": false }
pulse: 0.07196521759
{ "lastTime": 2.59523773193, "time": 2.66720294952, "pin": B12, "state": false }
pulse: -479.92019176483
{ "lastTime": -3452.90347862243, "time": -3932.82367038726, "pin": B12, "state": false }
pulse: -1103.95337390899
{ "lastTime": -2252.50851249694, "time": -3356.46188640594, "pin": B12, "state": false }
pulse: -95.96041870117
{ "lastTime": -3451.59917259216, "time": -3547.55959129333, "pin": B12, "state": false }
pulse: 3136.09915256500
{ "lastTime": -3130.59095287322, "time": 5.50819969177, "pin": B12, "state": false }
pulse: -2831.99997043609
{ "lastTime": 5.50820636749, "time": -2826.49176406860, "pin": B12, "state": false }
pulse: -207.89380645751
```

If the JS implementation is the only way I'll surely make a wrapper for it so it reads something like
```js
digitalPulseIn(BTN1, true, function(pulse) {
  console.log("got pulse of " + pulse + " ms");
});
```

If you're certain that this is a no-go in the c side of Espruino - feel free to delete/decline the issue :) We could continue this discussion on the forums if you'd prefer.

Thanks.

gfwilliams, 2015-01-21:
Accuracy should be around 1uS, as it uses a high speed timer to get the current time. Are you using the Espruino board? the values you're getting look very strange though - I haven't come across that before at all. Are you using one of my builds, or one of your own? It could be some issue with the compiler.

For Ultrasonic range sensors, does the module here not work? http://www.espruino.com/HC-SR04

Or if you use:

```
setWatch(A1, function(e) {
  var d = e.time - e.lastTime;
  // ...
}, { repeat: true, edge:"falling" });
```

It only has to call into JS code once (lastTime is still set to the time of the rising edge).

It still doesn't help with the time that's reported back. Can you also add a `console.log(getTime())` to your code to help to see if it's the global timer that is misbehaving?

AlexanderBrevig, 2015-01-27:
Unable to reproduce with both the Espruino flashed from Web IDE and the current version in the repository.
It seems setWatch will work just fine! Feel free to close this and thanks for pointing me to a good alternative.

gfwilliams, 2015-01-28:
Great - thanks for the update!

closed 2015-01-28 by gfwilliams