espruino/Espruino #489: JSON.stringify to dump types arrays as objects

gfwilliams, 2015-02-02:
https://github.com/espruino/Espruino/commit/17b6c3f333acf9b1630a979ac2602c240c57e18a#commitcomment-9507787

```
JSON.stringify(new Uint8Array([0,1])); // Produces the string: '{"0":0,"1":1}'
```

Should keep the same output format as currently when it's used to print out to the console though.

gfwilliams, 2016-03-17:
As in the bug above, dumping `new Uint8Array(...)` isn't valid JSON at all

SpenceKonde, 2016-03-17:
Hmm - the standard way is hideous. Wouldn't projecting it as a normal array be much better?

gfwilliams, 2016-03-17:
Yes, but then it's not doing what JSON.stringify should, and someone will complain that it's not compliant.

So yeah, this is why I've been putting off fixing it :)

gfwilliams, 2016-10-13:
Meh. Dumping as non-spec-compliant arrays anyway, but documenting it in `JSON.stringify`

commit ac94d4f8 (2016-10-13): JSON.stringify now dumps typed arrays as arrays (fix #489)
commit 66e7168b (2017-02-16): JSON.stringify now dumps typed arrays as arrays (fix #489)
closed 2016-10-13 by gfwilliams