espruino/Espruino #493: SPI.send sporadically sending incorrect output

bretg, 2015-02-12:
Working on an app that sends SPI data to a string of LEDs. Starting with v72, there's a sporadic flicker in one or more of the LEDs. Most of the time the flicker is of a color previously sent, which led me to suspect the SPI driver -- a hardware problem would probably show as random colors. 

So I worked backwards through the recent SPI module changes and found the flicker is not present in v71. Diffed jswrap_spi_i2c.c between 71 and 72 -- not many changes and nothing jumped out at me. 72 works _most_ of the time, but at some random interval, a light that changes flickers before the next update. Maybe some kind of memory mutex when there's another activity taking place? The issue seems to surface more often when the program is generally busier.

Here's a minimal program that displays the problem for 4 LEDs:

function loop() {
    SPI2.send4bit([0,32,0,32,32,32,32,32,32,32,32,32], 1, 3);
    // the lights are updated in a loop - even though it's sending red for the first GRB data,
    // the light occasionally flickers white for a cycle before returning to red the next cycle
}
SPI2.setup({baud:3200000, mosi:B15});
SPI2.send4bit([32,32,32,32,32,32,32,32,32,32,32,32], 1, 3);  // all white
SPI2.setup({baud:3200000, mosi:B15});  // tried with and without set between each call
SPI2.send4bit([0,32,0,32,32,32,32,32,32,32,32,32], 1, 3); // first LED red
setInterval(loop,250);

(FWIW - looking through the code and change logs reinforces how much work has gone into this product. Thanks.)

gfwilliams, 2015-02-12:
What happens if you do:

```
SPI2.send4bit(new Uint8Array([32,32,32,32,32,32,32,32,32,32,32,32]), 1, 3);
```

And are you using the official builds on the Espruino board?

My guess is it's actually just a delay between bytes coming out of SPI, rather than invalid data. The WS2811s are very sensitive to timing, and any gaps in the stream of bytes might cause glitches.

If you had an oscilloscope you could set up the SCK line on SPI and check it for gaps.

It's actually 'on the edge' at the moment, so any changes - for instance slightly slower array accesses - could mean occasional gaps. It's why 16 bit SPI sends are used and IRQs disabled - otherwise the SPI output would definitely end up with gaps in it.

I guess there are a few potential fixes:

* Attempt to speed up the 'send4bit' code
* Attempt to speed up the array iterator code
* Use DMA in a circular buffer for SPI sends - that'd be my preference as it would speed up everything, potentially even allowing asynchronous SPI. There's already been some work done on this for the TV output code, but it'd need quite a bit of work to get it running on any SPI port.

bretg, 2015-02-13:
> And are you using the official builds on the Espruino board?

Yes - official builds, official board (1.3).

> My guess is it's actually just a delay between bytes coming out of SPI, rather than invalid data.

Yes, this could be right. Will study the WS2812b and gather evidence over the weekend, though not likely from a scope.

So, the "new Uint8Array" approach did work to eliminate the flicker. (Why?)  But at the cost of steadily rising memory. Will try to find a way to make this work using the same piece of memory.

gfwilliams, 2015-02-13:
Ok, that's a relief then. On the WS2811 page I do try and encourage people to use the Uint8(Clamped)Arrays.

You might want to see http://www.espruino.com/Performance - basically if you use a normal array, it's a 'sparse' array - with an Index and a Value stored for each element. The value can be anything, which means accesses are slower as the value has to be checked and then converted if needed.

To add to that, in order to save RAM the index and value are often packed together (if possible), so that's yet another check that's needed.

It's not normally that big a deal, but when it's got to push the data out so quickly it looks like the added overhead just pushes it over the edge sometimes.

gfwilliams, 2015-02-13:
Ok, I've just pushed in some code that should improve the speed of array accesses where the array contains small ints. Hopefully that'll help.

By the way:  

> But at the cost of steadily rising memory. 

Do you have an example? There shouldn't be any memory leaks.

When you've got a (big?) buffer like for the LEDs it's good practice to keep it hanging around though - it'll take a little while to allocate each time otherwise.

closed 2015-02-13 by gfwilliams

bretg, 2015-02-14:
Can't reproduce the process.memory() growth anymore - user error probably.

Just re-read your WS2811 page - did run across it a few months ago but forgot a key lesson. Partly sorry to have bothered you, partly pleased with good support. Thanks.

bretg, 2015-02-14:
Found the memory problem I'd run into -- at one point I'd tried doing the SPI2.setup() on every loop to see if that might work. That function seems to takes a block on every call that doesn't get released.

gfwilliams, 2015-02-14:
Thanks!  I'll look into it.  It should be easy to fix if there's a leak there

commit 96f13167 (2015-02-13): Improve jsvIteratorGetIntegerValue speed for arrays (fix #493)