espruino/Espruino #494: linux build fails

bartmichu, 2015-02-13:
Hi, I'm not able to build the latest thing (git clone) under linux:

```
CC src/jslex.o
In file included from src/jslex.h:18:0,
                 from src/jslex.c:14:
src/jsvariterator.h: In function ‘jsvStringIteratorFree’:
src/jsvar.h:261:20: error: inlining failed in call to always_inline ‘jsvUnLock’: function body not available
 ALWAYS_INLINE void jsvUnLock(JsVar *var);
                    ^
In file included from src/jslex.h:19:0,
                 from src/jslex.c:14:
src/jsvariterator.h:111:3: error: called from here
   jsvUnLock(it->var);
   ^
Makefile:1160: recipe for target 'src/jslex.o' failed
make: *** [src/jslex.o] Error 1
```

Any ideas?

gfwilliams, 2015-02-13:
Do you maybe have an old compiler? 

Worst case you can comment out ALWAYS_INLINE in jsutils.h, but it is needed for compact ARM builds

bartmichu, 2015-03-22:
Sorry for a late reply, I missed this one.

So I'm using up-to-date version of Fedora 21. Tried to comment it out but to no avail, I guess I'm doing something wrong as my C knowledge is null.

gfwilliams, 2015-03-23:
[on this line](https://github.com/espruino/Espruino/blob/master/src/jsutils.h#L187) try changing:

```
#if !defined(SAVE_ON_FLASH) && !defined(DEBUG)  && !defined(NO_ALWAYS_INLINE) && !defined(__MACOSX__)
// Don't do this on Mac because no need, and clang doesn't like the inline + attribute (at least according to c05b5e701a118ec0b7146f2a6bb8c06a26d0652c)
#define ALWAYS_INLINE inline __attribute__((always_inline))
#else
#define ALWAYS_INLINE inline
#endif
```

to

```
#define ALWAYS_INLINE inline
```

bartmichu, 2015-03-23:
It produces lots of warnings now but works! Thanks Gordon.

gfwilliams, 2015-04-14:
I think I may have fixed this now by not attempting to inline on Linux at all (We'll just assume the compiler knows best :) )

closed 2015-04-14 by gfwilliams

bartmichu, 2015-04-16:
I can confirm that it builds now on the same system so looks like it's fixed, thanks!