espruino/Espruino #495: Same name for multiple properties

bartmichu, 2015-02-13:
I believe it shouldn't be possible to create two properties with the same name:

```
>var foo = {bar: 1, baz: 2, bar: 3}
={ "bar": 1, "baz": 2, "bar": 3 }
```

But then:
```
>for (var i in foo) {console.log(foo[i])};
1
2
1
```

After playing with other interpreters I would expect `{"bar": 3, "baz": 2}` but I'm not sure what's in the specs.

gfwilliams, 2015-02-13:
Ahh.  Thanks... Yes,  I think this should probably report an error.

gfwilliams, 2015-06-03:
Seems like other interpreters don't report an error, so not doing it here either.

closed 2015-06-03 by gfwilliams

bartmichu, 2015-06-03:
Error reporting is not an issue, I was thinking more about returned value i.e. that second "bar" property shouldn't even be added to the object.

commit 3c994464 (2015-06-03): Duplicate properties in an object defn. now cause second to be used (fix #495)