espruino/Espruino #502: Allow users on the Espruino website to contribute updated documentation

gfwilliams, 2015-03-04:
You can do this now, but it's more painful as it requires GitHub knowledge.

See suggestions here: http://forum.espruino.com/conversations/263655/#comment12136890

Not actually an Espruino bug - this needs implementing in the website.

gadicc, 2015-03-05:
Hey @gfwilliams, long time :>  By complete chance I popped by the repo today and saw this issue, while coincidentally busy working on the rewrite of [Meteorpedia](http://meteorpedia.com/read/Main_Page).  It's still some time away but I believe it could solve this issue.  The new version features realtime preview and collaborative live editing (like google docs), and of course all the existing features still relevant for here: categories like in wikipedia/mediawiki ('needs-work' etc), revision history, etc.

As for my espruino work, unfortunately I've had no time at all for further playing, as much as I enjoyed it... other projects simply taking up too much time, but if I can find another way to contribute I'll be happy :)

gfwilliams, 2015-03-09:
No problem - the issue is actually just feeding the edits back to GitHub in a way that I can keep track of them. I think given the number of edits, concurrent editing (or even having a 'nice' editor) isn't such a big deal.

It'd be great if someone could find a way to automate the generation of a GitHub PR, but I think this would probably have to be done server-side as a lot of users wouldn't have a GitHub account. I think realistically it's something I'll end up bodging up.

gfwilliams, 2017-06-08:
There's now a `=>` logo next to each function in the reference that links back to GitHub. I think that's probably the easiest, most sensible way to handle this

closed 2017-06-08 by gfwilliams