espruino/Espruino #510: arguments & callee not supported...

aplikatika, 2015-03-19:
I tried:
function a() { return arguments.callee; } 

with:
a()();

and failed with:
Uncaught [object Object]
 at line 1 col 19
{ return arguments.callee; }
                   ^
in function "a" called from line 1 col 3
a()()
   ^
 at line 1 col 3
a()()
   ^

I am sure we can fix it...

gfwilliams, 2015-03-20:
Thanks! It shouldn't be too hard to add.

However [according to Mozilla](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Functions/arguments/callee) this is disallowed in ES5 strict mode (so is effectively deprecated), because usually you can just use named functions.

I makes me wonder if we should actually add it? Do you have a use case where it's absolutely required?

opichals, 2015-03-20:
A 3rd party code you might want to run without changes (not using strict mode) would be a good reason to implement this IMO

gfwilliams, 2015-03-20:
Is there such third party code right now, or is it theoretical?

I ask because I only have limited development time, and it probably makes more sense for me to use it on features that people will use, rather than features that ECMA are actually trying to stop people using.

xk, 2015-03-20:
On 20/03/2015, at 14:04, Gordon Williams wrote:

> Is there such third party code right now, or is it theoretical?
> 
> I ask because I only have limited development time, and it probably makes more sense for me to use it on features that people will use, rather than features that ECMA are actually trying to stop people using.

+1

-- 
( Jorge )();

opichals, 2015-03-20:
Theoretical in my case. +1 for giving this a low priority.

gfwilliams, 2016-02-08:
Sorry - dropping this. It's basically deprecated and we've had almost a year with nobody caring about it :)

closed 2016-02-08 by gfwilliams