espruino/Espruino #512: ERROR: Got EOF expected '}'

rwaldron, 2015-03-30:
Reduced test case: 

```js
function f(a) {
  console.log(a);
}

f(1);
```

Result: 

```
>echo(0);
1
ERROR: Got EOF expected '}' at line 1 col 16
{  console.log(a)
               ^
in function "f" called from line 1 col 3
=undefined
> 
```

I was going to write a patch for this, but it building the espruino runtime no longer works on OSX.

gfwilliams, 2015-03-30:
Wow, that's strange. Just tried it here on Linux and on the Espruino board with latest firmware and can't reproduce. Are there some magic characters in what you write above that aren't coming out?

Also, can you report build errors in another bug? It should work...

gfwilliams, 2015-03-30:
Ok, just tried to build on Mac and it failed here too - I've now sorted it out

rwaldron, 2015-03-30:
Ah, it's possible that my update didn't work? I was trying to do all of this from the Chrome app IDE. I will run through again and see if I've missed anything.

rwaldron, 2015-03-30:
I just reran the flasher from the IDE, twice and it appears to have worked correctly. The error is no longer present.

closed 2015-03-30 by rwaldron

gfwilliams, 2015-03-30:
Maybe - see what it reports under `Board Information` on the first page of settings after you connect. It should say 1v75.

Could you still have the earliest version of the firmware that shipped on the KickStarter boards? That was 1v4something - I seem to recall that in a hurry to get the boards out, i'd made some really bad mistake and basically regressed the as-supplied firmware version to the point where it complained about the final bracket on functions.

gfwilliams, 2015-03-30:
Ahh, ok, that's almost certainly what it was then. Had you ever updated that board before?

rwaldron, 2015-03-30:
> Could you still have the earliest version of the firmware that shipped on the KickStarter boards? That was 1v4something - I seem to recall that in a hurry to get the boards out, i'd made some really [...]

...

> Had you ever updated that board before?

I have four now, and it's **likely** that I grabbed one of the previously unused ones from my newly re-organized hardware bench—whoops ;)