espruino/Espruino #513: this isn't propagated to eval()

SpenceKonde, 2015-03-31:
```

var t={};
t.test=function(){console.log(this);eval("console.log(this);")};
t.test();

```

produces:

```

>t.test();
{
  "test": function () {console.log(this);eval("console.log(this);")}
 }
{
  "t": {
    "test": function () {console.log(this);eval("console.log(this);")}
   },
  "SyntaxError": function () { [native code] },
  "console": function () { [native code] }
 }
=undefined
>

```

should produce:

```

>t.test();
{
  "test": function () {console.log(this);eval("console.log(this);")}
 }
{
  "test": function () {console.log(this);eval("console.log(this);")}
 }
=undefined
>

```

(At least that's how it works in browser js consoles);

gfwilliams, 2015-03-31:
It turns out this is particularly nasty. If you do a 'normal' function call it behaves like you'd expect:

```
var t={};
function f() { console.log(this); }
t.test=function(){console.log(this);f();eval("console.log(this);")};
t.test();
```

```
{"test": function (){console.log(this);f();eval("console.log(this);")}}
[object Window]
{"test": function (){console.log(this);f();eval("console.log(this);")}}
```

So it's a special case for `eval` - but it gets worse: http://stackoverflow.com/questions/9107240/1-evalthis-vs-evalthis-in-javascript

`(1,eval)('this')` is different to `eval('this')`. I'm not implementing that one though, because it's a bit nuts and looks like it's really painful to do at the moment.

closed 2015-03-31 by gfwilliams

SpenceKonde, 2015-03-31:
I wasn't aware of the (1,eval)('this') - IMO, that's just perverse.

My understanding of eval() is that, without doing weird things, it's supposed to evaluate the code within the current context. That's what makes it such an incredible tool.

commit bf40f34e (2015-03-31): Propagate `this` into eval (fix #513)