espruino/Espruino #520: Add 'module' variable for modules

gfwilliams, 2015-04-17:
http://forum.espruino.com/comments/12223159/

This should work - currently you need to use just `exports`:

```
// the module
module.exports = function () {
  console.log('I am the module');
};
// the consumer
var module = require('my-module');
module(); // I am the module
```

gfwilliams, 2015-04-17:
And also ensure that `this` is set correctly: http://forum.espruino.com/comments/12223222/

WebReflection, 2015-04-17:
please bear in mind that reassigning `exports` does not actually export a thing in node.js, that is why people use `module.exports = ...` instead.

```js
exports = function wut() {};
```
If we require that file in node the exported value would be just a regular empty object.

In order to actually overwrite the `exports` you need to overwrite the module property.

```js
module.exports = function wut() {};
```

Inside a module, and only when required, `exports` is the equivalent of `this`

However, in the main file, `this` will be the `global` object instead.
An old trick to know if a file is loaded as module is indeed:
```js
function log(s) {
  console.log(s);
}
if (this === global) {
  log('main');
} else {
  module.exports = log;
}
```

WebReflection, 2015-04-17:
As summary, this is how in JS you would evaluate a module:

```js
function evalRequire(scriptContent) {
  var exports = {};
  var module = {exports: exports};

  Function(
    'global', 'exports', 'module',
    scriptContent
  ).call(
    exports,
    global,
    exports,
    module
  );

  return module.exports;
}
```

As you can see, replacing the `exports` variable inside the module won't actually affect the exported value, since that's retrieved through the `module` and `exports` is simply a shortcut.

`node.js` has been like this since version `0.4` if I remember correctly, hope this helps.

gfwilliams, 2015-04-17:
Thanks, yes, it does. I'm actually pretty sure I added the `exports=` behaviour at someone's request. Looks like I should have double-checked first!

commit 6a610385 (2015-04-23): Add 'modules' variable, and set 'this' to 'exports' when parsing a module (fix #520)
closed 2015-04-23 by gfwilliams