espruino/Espruino #528: res.on('close') doesn't get called for pages w/no body (ie, REST interfaces!)

SpenceKonde, 2015-04-30:
This is a serious issue, as it makes it impossible to use the ESP8266 wifi module with some REST APIs. 

The res.on('close') callback seems to be ignored if the response has no body. For example, i send a GET request for http://192.168.2.14/api/relay/1/on

This returns a response with status code 200 OK and the following headers:

Access-Control-Allow-Origin: *
Cache-Control: max-age=600
Connection: close

and no page content (0 bytes)

In this case, res.on('close') is never called, and so the memory used for the request is never freed.

gfwilliams, 2015-05-01:
Ahh - that's what was causing your problem?

So `Unlink` is still sent by the ESP8266 in that case?

gfwilliams, 2015-05-01:
Luckily this can be reproduced even on a linux build, so hopefully I'll have a fix for you by the end of today

commit 5ae24206 (2015-05-01): Ensure res.on('close') is called for empty HTTP requests (fix #528)
closed 2015-05-01 by gfwilliams