espruino/Espruino #531: Teensy 3.1 support

Jorgen-VikingGod, 2015-05-05:
hi guys,
I really would like to have a **Teensy 3.1** support of Espruino.
A port of Espruino to run JavaScript on Teensy would be awesome, because the Teensy board is fast and powerful and I love to play with ;)

## Specs:
http://www.pjrc.com/teensy/teensy31.html#specs
<table align="center" width="680" cellspacing="3" cellpadding="6">
<tbody><tr class="top"><th>Feature</th><th>Teensy 3.0</th><th>Teensy 3.1</th><th>Units</th>

</tr><tr class="odd">
<td>Price</td>
<td align="center">19.00</td>
<td align="center">19.80</td>
<td>US Dollars</td>
</tr>

<tr class="odd">
<td>Processor<br>&nbsp;&nbsp;&nbsp;&nbsp;Core<br>&nbsp;&nbsp;&nbsp;&nbsp;Rated Speed<br>&nbsp;&nbsp;&nbsp;&nbsp;Overclockable</td>
<td align="center"><small>MK20DX128VLH5</small><br>Cortex-M4<br>48<br>96</td>
<td align="center"><small>MK20DX256VLH7</small><br>Cortex-M4<br>72<br>96</td>
<td><br><br>MHz<br>MHz</td>
</tr>

<tr class="odd">
<td>Flash Memory<br>&nbsp;&nbsp;&nbsp;&nbsp;Bandwidth<br>&nbsp;&nbsp;&nbsp;&nbsp;Cache</td>
<td align="center">128<br>96<br>32</td>
<td align="center">256<br>192<br>256</td>
<td>kbytes<br>Mbytes/sec<br>Bytes</td>
</tr>

<tr class="odd">
<td>RAM</td>
<td align="center">16</td>
<td align="center">64</td>
<td>kbytes</td>
</tr>

<tr class="odd">
<td>EEPROM</td>
<td align="center">2</td>
<td align="center">2</td>
<td>kbytes</td>
</tr>

<tr class="odd">
<td>Direct Memory Access</td>
<td align="center">4</td>
<td align="center">16</td>
<td>Channels</td>
</tr>

<tr class="odd">
<td>Digital I/O<br>&nbsp;&nbsp;&nbsp;&nbsp;Voltage Output<br>&nbsp;&nbsp;&nbsp;&nbsp;Voltage Input</td>
<td align="center">34<br>3.3V<br>3.3V Only</td>
<td align="center">34<br>3.3V<br>5V Tolerant</td>
<td>Pins<br>Volts<br>Volts</td>
</tr>

<tr class="odd">
<td>Analog Input<br>&nbsp;&nbsp;&nbsp;&nbsp;Converters<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Resolution<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Usable<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Prog Gain Amp<br>&nbsp;&nbsp;&nbsp;&nbsp;Touch Sensing<br>&nbsp;&nbsp;&nbsp;&nbsp;Comparators</td>
<td align="center">14<br>1<br>16<br>13<br>0<br>12<br>2</td>
<td align="center">21<br>2<br>16<br>13<br>2<br>12<br>3</td>
<td>Pins<br><br>Bits<br>Bits<br><br>Pins<br><br></td>
</tr>

<tr class="odd">
<td>Analog Output<br>&nbsp;&nbsp;&nbsp;&nbsp;DAC Resolution</td>
<td align="center">0<br>-</td>
<td align="center">1<br>12</td>
<td>Pins<br>Bits</td>
</tr>

<tr class="odd">
<td>Timers<br>&nbsp;&nbsp;&nbsp;&nbsp;FTM Type<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;PWM Outputs<br>&nbsp;&nbsp;&nbsp;&nbsp;PDB Type<br>&nbsp;&nbsp;&nbsp;&nbsp;CMT (infrared) Type<br>&nbsp;&nbsp;&nbsp;&nbsp;LPTMR Type<br>&nbsp;&nbsp;&nbsp;&nbsp;PIT (interval) Type<br>&nbsp;&nbsp;&nbsp;&nbsp;Systick<br>&nbsp;&nbsp;&nbsp;&nbsp;RTC (date/time) **</td>
<td align="center">11 Total<br>2<br>10<br>1<br>1<br>1<br>4<br>1<br>1</td>
<td align="center">12 Total<br>3<br>12<br>1<br>1<br>1<br>4<br>1<br>1</td>
<td valign="top"><br><br>Pins</td>
</tr>

<tr class="odd">
<td>Communication<br>&nbsp;&nbsp;&nbsp;&nbsp;USB<br>&nbsp;&nbsp;&nbsp;&nbsp;Serial<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;With FIFOs<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;High Res Baud<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Fast Clock<br>&nbsp;&nbsp;&nbsp;&nbsp;SPI<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;With FIFOs<br>&nbsp;&nbsp;&nbsp;&nbsp;I2C<br>&nbsp;&nbsp;&nbsp;&nbsp;CAN Bus<br>&nbsp;&nbsp;&nbsp;&nbsp;I2S Audio<br>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;FIFO Size</td>
<td align="center"><br>1<br>3<br>1<br>3<br>2<br>1<br>1<br>1<br>0<br>1<br>4</td>
<td align="center"><br>1<br>3<br>2<br>3<br>2<br>1<br>1<br>2<br>1<br>1<br>8</td>
<td></td>
</tr>

<tr class="odd">
<td></td>
<td align="center"></td>
<td align="center"></td>
<td></td>
</tr>

<tr class="odd">
<td colspan="4"><small><a name="rtc">**</a> RTC requires a 32.768 kHz crystal &amp; 3V battery.  See the
<a href="td_libs_Time.html#teensy3">Time Library</a> for details.</small></td>
</tr>

</tbody></table>

## Pinout:
http://www.pjrc.com/teensy/pinout.html
[image]
[image]

gfwilliams, 2015-05-05:
Thanks - anyone is welcome to follow the instructions in the Readme and try and make their own build of Espruino for the Teensy. However because the ARM is from a different manufacturer, porting all the I2C/SPI/Serial libraries over will be quite a bit of work.

Personally I don't have the time for it though. If you want a small board that runs Espruino there's the [Espruino Pico](http://www.espruino.com/Pico) - which is actually smaller, faster, and has more memory. Money from the sales of it also pays for me to keep developing and supporting the Espruino firmware :)

Jorgen-VikingGod, 2015-05-05:
Hey that's awesome - I did not see the Pico before.
But where can I buy one? Currently entered my mail to the wait list - is there a possibility to get one earlier to play with. Perhaps doing a port of FastLED library would be great on Pico as well ;)
I like Teensy, because it is fast working with FastLED - and I do a lot with LEDs...
If you have some details, please mail me (JorgenVikingGod@gmail.com)

gfwilliams, 2015-05-06:
Well, you don't really need the FastLED library - Espruino works pretty well with reasonable quantities of LEDs, for instance: http://www.espruino.com/WS2811

You can do strips with simple arrays, or you can even handle big grids of LEDs like http://www.espruino.com/RGB123 with the built in Graphics library.

Unfortunately at the moment I'm waiting for more stock of the Espruino Pico (which could be almost a month away). I've got a whole lot of boards that had one edge slightly damaged in production (other than that they work perfectly), and I'll try and get those online to sell in the next week or so - I'll send an e-mail to all those who put their name on the list when they're available.

gfwilliams, 2022-11-02:
This is 7 years old and I don't think you can even buy Teensy 3 now - closing!

closed 2022-11-02 by gfwilliams