espruino/Espruino #532: misleading setTimeout and setInterval

WebReflection, 2015-05-05:
[Standard timers](http://www.w3.org/TR/2011/WD-html5-20110525/timers.html#timers) have the ability to pass arbitrary amount of arguments since about ever and in every JS engine.

This should log 123, and not an object with a time property.

```js
setTimeout(function (i) { console.log(i); }, 500, 123);
```

This causes incompatibilities with many nodejs modules, and forces creation of user-land callbacks even when it's not necessary.

Moreover, the current passed object is totally unexpected and also the length of the arguments is 2, with an undefined as second argument, which is also inconsistent and unexpected.

gfwilliams, 2015-05-05:
Odd - I was totally unaware of that. Definitely looks like a good thing to add.

gfwilliams, 2015-05-06:
OOps - referenced wrong issue

WebReflection, 2015-05-06:
I was wondering ... indeed there was no arguments related code in the fix ... oh well, good you fixed the other one anyway. Cheers

commit 4655fc47 (2015-05-06): Make bind reference function internals not copy them. Fix scoped vars in bind (fix #532)
commit 3f59263c (2015-05-06): Use jsvUnlockMany to tidy up code and save some space
            jsiExecuteEventCallback cal now take an arbitrary number of arguments
            Allow setTimeout/setInterval to take extra arguments (fix #532)
closed 2015-05-06 by gfwilliams