espruino/Espruino #533: Soped reference lost after bound setTimeout

WebReflection, 2015-05-06:
This is quite nasty ... the `t` reference goes nuts if you use bind for the `setTimeout` or setInterval

```js
var obj = {
  timer: function (s, unitms) {
    var t = 0;
    if (!unitms) unitms = 300;
    return {
      times: 0,
      start: function start() {
        this.times++;
        console.log(t);
        t = setTimeout(function () {
          t = setTimeout(start.bind(this), 1000);
        }.bind(this), 1000);
        return this;
      },
      stop: function () {
        console.log(t);
        clearTimeout(t);
        return this;
      }
    };
  }
};

var signal = null;

function watch() {
  if (signal === null) {
    signal = obj.timer();
    signal.start(write);
  } else {
    console.log('times: ' + signal.times);
    signal.stop();
    signal = null;
  }
}

function onInit() {
  clearWatch();
  setWatch(
    watch,
    BTN1,
    {
      repeat: true,
      debounce: 50,
      edge: 'rising'
    }
  );
}
```

once `onInit()` is invoked, the expected behavior is that when the button is pressed, it starts, looping (no matter what it does now, it's just for this bug purpose that I am logging `t`)

Once it's pressed again, the timeout should be cleared through the t variable which has never assigned to anythig different than a timer.

Please note that `signal.times` keep increasing without problems

WebReflection, 2015-05-06:
forgot to mention, I am testing on a Pico

WebReflection, 2015-05-06:
to simplify the test, write `watch()` then try to write `watch()` again and see that `t` is always `0` . The `setTimeout` is related to the returned object method, the `t` variable should NOT be lost in the process. You can test with any other engine too and see things are just fine.

```js
var obj = {
  timer: function () {
    var t = 0;
    return {
      times: 0,
      start: function start() {
        this.times++;
        console.log(t);
        t = setTimeout(start.bind(this), 1000);
        return this;
      },
      stop: function () {
        console.log(t);
        clearTimeout(t);
        return this;
      }
    };
  }
};

var signal = null;

function watch() {
  if (signal === null) {
    signal = obj.timer();
    signal.start();
  } else {
    console.log('times: ' + signal.times);
    signal.stop();
    signal = null;
  }
}
```

Thanks for checking this out

gfwilliams, 2015-05-06:
Thanks - that is a strange one. Printing the value of `t` after `setTimeout` implies that it's getting set to the correct thing, but then getting lost somehow.

Also, you can check what the current timers are using `Object.keys(global[">"].timers)` - but that's just an internal thing so may change in the future

WebReflection, 2015-05-06:
please note if you remove the `bind` everything is fine

```js
var obj = {
  timer: function () {
    var t = 0;
    return {
      times: 0,
      start: function start() {
        console.log(t);
        t = setTimeout(start, 1000);
        return this;
      },
      stop: function () {
        console.log(t);
        clearTimeout(t);
        return this;
      }
    };
  }
};

var signal = null;

function watch() {
  if (signal === null) {
    signal = obj.timer().start();
  } else {
    signal.stop();
    signal = null;
  }
}
```

It looks like the binding re-evaluate the entire scope and loose the original `t` reference in the outer one (whenever this makes sense or not)

gfwilliams, 2015-05-06:
Simpler example:

```
function timer() {
    var t = 0;
    return {
      start: function() {
        console.log(t++);
        t = setTimeout(this.start.bind(this), 1000);
      }
    };
  }

var signal = timer();
signal.start();
```

This goes:

```
0
1
1
1
1
```

So while `bind` is causing it, the value of t gets read (and written to what would appear to be the same variable) but somehow doesn't get stored!

gfwilliams, 2015-05-06:
By the way, what do you mean by 'Soped'?

WebReflection, 2015-05-06:
I mean **typo** ... it supposed to be **Scoped** and not **Soped** ... dho!

gfwilliams, 2015-05-06:
Closed by https://github.com/espruino/Espruino/commit/4655fc47b24147adf8c7ecc673846015ca386b67

closed 2015-05-06 by gfwilliams