espruino/Espruino #539: Junk characters appearing in serial when printing huge amounts of serial data

SpenceKonde, 2015-05-28:
pin A8 connected to 433mhz RF receiver. Serial output periodically contains garbage. 

```

var rxpin = A8;
var n="";
var wf;

function startListen() {
	wf=setWatch(sigOff,rxpin,{repeat:true,edge:"falling"});
	console.log("Listening started");
}

function stopListen() {
	if (wf) {clearWatch(wf);}
	wf=0;
	n="";
  console.log("listening stopped");
}

function sigOff(e) {
	var d=e.time-e.lastTime;
  console.log(d);
	if (d>0.00012 && d<0.00075) n+=d>0.000425?1:0;
	else n="";
	if (n.length==z) parseRx(n);
}

setTimeout("stopListen()",5000);startListen(); 

```

Example of output:

```

0.00059604644
0.00071525573
0.00017261505
0.00023841857
0.000120	¨È}äÎ 4Î ñQ 	Pñÿÿÿ096225738
0.00003623962
0.00010871887
0.00087356567
0.00102317333
0.00084018707
0.00006008148

```

Timing seems to be random. Sometimes as little as ~10 readings between them, other times you can scroll past a few screens without seeing one.

gfwilliams, 2015-05-28:
Which board is this on? The USB implementation is different on the 2 boards...

I think I actually found the problem yesterday while working on USB HID. The USB code calls a function with a buffer that was allocated on the stack, and it turns out that the function being called just passes the pointer straight to the USB hardware... So if you're sending data *and* executing code at the same time, in some cases that data gets overwritten with stack contents before it can be transmitted.

I'm actually replacing the whole USB implementation as part of  https://github.com/espruino/Espruino/issues/221, so hopefully this will end up getting fixed as part of that...

SpenceKonde, 2015-05-28:
Sorry, this was on the Pico.

gfwilliams, 2015-05-28:
Thanks - hopefully this'll get solved soon then :)

gfwilliams, 2015-06-05:
I'd be interested to see if this happens on the USB HID firmware: http://forum.espruino.com/conversations/268671/#comment12319741

USB has been totally rewritten, so you'd hope not.

gfwilliams, 2015-08-10:
1v80 should have this fixed. @SpenceKonde please could you re-open this if you find there are still problems? However I found a bug that would have caused exactly this issue while doing the USB HID work, so I'm hopeful it's fixed now.

closed 2015-08-10 by gfwilliams