espruino/Espruino #555: NUCLEO-F411RE target fails building

siorpaes, 2015-08-03:
Hello,

I am trying to build Espruino for NUCLEO-F411RE target using

`
NUCLEOF411RE=1 RELEASE=1 V=1 make
`

 but stumbling in the following error:

`
arm-none-eabi-gcc -Wall -Wextra -Wconversion -Werror=implicit-function-declaration -O3 -fno-common -fno-exceptions -fdata-sections -ffunction-sections -flto -fno-fat-lto-objects -Wl,--allow-multiple-definition --param inline-unit-growth=6 -c -mlittle-endian -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DGIT_COMMIT=c2f2e27f2bdf3623a011f6ec22ec7f8190c9cf71 -DNO_ASSERT -DRELEASE -DBUILDNUMBER="1" -DUSE_MATH -DUSE_GRAPHICS -DUSE_NET -DUSE_CC3000 -DSEND_NON_BLOCKING -DSTM32F4 -DARM -DLINK_TIME_OPTIMISATION -DFAKE_STDLIB -DSTM32 -DUSE_STDPERIPH_DRIVER=1 -DSTM32F411RET6 -DNUCLEOF411RE -DSTM32F411xE -DEMBEDDED -I/home/ohrm/Espruino -I/home/ohrm/Espruino/targets -I/home/ohrm/Espruino/src -I/home/ohrm/Espruino/gen -I/home/ohrm/Espruino/libs/math -I/home/ohrm/Espruino/libs/graphics -I/home/ohrm/Espruino/libs/network -I/home/ohrm/Espruino/libs/network -I/home/ohrm/Espruino/libs/network/http -I/home/ohrm/Espruino/libs/network/js -I/home/ohrm/Espruino/libs/network/cc3000 -I/home/ohrm/Espruino/targetlibs/stm32f4 -I/home/ohrm/Espruino/targetlibs/stm32f4/lib -I/home/ohrm/Espruino/targetlibs/arm -I/home/ohrm/Espruino/targets/stm32 -DGIT_COMMIT=c2f2e27f2bdf3623a011f6ec22ec7f8190c9cf71 -DNO_ASSERT -DRELEASE -DBUILDNUMBER="1" -DUSE_MATH -DUSE_GRAPHICS -DUSE_NET -DUSE_CC3000 -DSEND_NON_BLOCKING -DSTM32F4 -DARM -DLINK_TIME_OPTIMISATION -DFAKE_STDLIB -DSTM32 -DUSE_STDPERIPH_DRIVER=1 -DSTM32F411RET6 -DNUCLEOF411RE -DSTM32F411xE -DEMBEDDED targetlibs/stm32f4/lib/system_stm32f4xx.c -o targetlibs/stm32f4/lib/system_stm32f4xx.o
targetlibs/stm32f4/lib/system_stm32f4xx.c: In function 'SetSysClock':
targetlibs/stm32f4/lib/system_stm32f4xx.c:522:29: error: 'PLL_N' undeclared (first use in this function)
     RCC->PLLCFGR = PLL_M | (PLL_N << 6) | (((PLL_P >> 1) -1) << 16) |
                             ^
targetlibs/stm32f4/lib/system_stm32f4xx.c:522:29: note: each undeclared identifier is reported only once for each function it appears in
targetlibs/stm32f4/lib/system_stm32f4xx.c:522:46: error: 'PLL_P' undeclared (first use in this function)
     RCC->PLLCFGR = PLL_M | (PLL_N << 6) | (((PLL_P >> 1) -1) << 16) |
                                              ^
make: *** [targetlibs/stm32f4/lib/system_stm32f4xx.o] Error 1
`

Note that the NUCLEOF401RE build perfectly.

Any hint/suggestion/quick fix ?

Many thanks in advance

David

gfwilliams, 2015-08-03:
Try now? Looks like the F411 was a bit of a hack (compiled as if it was an F401) and when I changed the STM32 libraries over I must have not realised and tried to do something specific.

There might be a better fix, but this compiles the binary in the same way that it always used to be compiled.

closed 2015-08-03 by gfwilliams

siorpaes, 2015-08-03:
Thanks! works perfectly now ;)

commit 3f8df706 (2015-08-03): fix F411 build (fix #555)