espruino/Espruino #556: Increase network buffer size

gfwilliams, 2015-08-03:
See http://forum.espruino.com/conversations/272025/

This is currently 64, but could hopefully be much bigger. Perhaps for sending we could even use `aalloc` to allocate just what we need.

Only thing to beware of is unless the CC3000 build gets massive static buffers put in too, it'll crash when we try and send more than 64 bytes (so should probably limit the max size in that case).

nkolban, 2015-10-08:
As discussed in the forum thread, sending data in 64 byte chunks is as thought slow.  On the ESP8266 board, it is quite noticable.

gfwilliams, 2015-10-08:
... although on ESP8266 you're allocating separate buffers - so for instance if we moved to 1500 bytes, that's presumably 3kB of RAM that you've now had to use up, which drastically reduces the amount of JS variables you can safely have.

nkolban, 2015-10-08:
I'm assuming that when `net_<board>_send()` is called then a buffer of data is passed in (say 1500 bytes).  Since this is stack allocated, ESP8266 duplicates it creating 3000 bytes in total.  However, almost immediately `net_<board>_send()` returns and the initial 1500 bytes of stack is released resulting in the necessary 1500 bytes being held for transmission.   Subsequent requests to send are immediately told "not yet" through a return code of 0 indicating that no consumption is possible just now.

What about making the buffer size a runtime configurable system property and allow the user to decide?

gfwilliams, 2015-11-26:
Just a note here, it seems the ESP8266 build has a max packet size of 544 bytes, so that is all we have to care about.

Some discussion here: http://forum.espruino.com/conversations/278181/#comment12649305

gfwilliams, 2016-02-08:
Already fixed: https://github.com/espruino/Espruino/commit/34326d03b9903efdaf6133cda626d49460ee744d

closed 2016-02-08 by gfwilliams