espruino/Espruino #559: Uncaught InternalError: setDeviceClockCmd: Unknown Device 816 with 1v80, not 1v79

asez73, 2015-08-09:
I just updated my Espruino board 1v4 to interpreter 1.80.
I tried to run a Serial4 logger and I received this error message from the very first line of code:

```
>reset();
=undefined
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v80 Copyright 2015 G.Williams
>Serial4.setup(115200,{tx:C10,rx:C11});
=undefined
Uncaught InternalError: setDeviceClockCmd: Unknown Device 816
 at line 1 col 37
Serial4.setup(115200,{tx:C10,rx:C11});
                                     ^
>dump()
Serial4.setup(115200, {"tx":C10,"rx":C11});
pinMode(C11,"input_pullup");
=undefined
> 

```

Interestingly it show C11 as 

> pinMode(C11,"input_pullup");

And furthermore, not copy / pastable, I had to type a Carriage return on the left console to get the '>' prompt immediatly after the error happened.

Version 1v79 was working perfectly just a few seconds before...

Removing the pin's defintions does give the same error's message.

```
>reset();
=undefined
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v80 Copyright 2015 G.Williams
>Serial4.setup(115200);
=undefined
Uncaught InternalError: setDeviceClockCmd: Unknown Device 816
 at line 1 col 21
Serial4.setup(115200);
                     ^
>dump()
Serial4.setup(115200);
pinMode(C11,"input_pullup");
=undefined
> 
```
And removing only rx's pin definition does not show the 
```
pinMode(C11,"input_pullup");
```

  after a dump() but the error remains.

```
>reset();
=undefined
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v80 Copyright 2015 G.Williams
>Serial4.setup(115200,{tx:C10});
=undefined
Uncaught InternalError: setDeviceClockCmd: Unknown Device 816
 at line 1 col 30
Serial4.setup(115200,{tx:C10});
                              ^
>dump()
Serial4.setup(115200, {"tx":C10});
=undefined
> 
```

asez73, 2015-08-09:
Changing Serial4 in Serial3 with the same pins did not show any error. However Serial3 doesn't receive any data.

asez73, 2015-08-09:
CHanging to Serial2 worked with A2 and A3 pins...

gfwilliams, 2015-08-10:
Thanks - Sorry about that. Just fixed this (so it'll work in the build from http://www.espruino.com/binaries/git/commits/594759be79485dc9ace2b8459fb99117f8c6407c) but there are still issues with Serial3 on C10/C11 (it works on B10/11) so leaving this bug open for now.

asez73, 2015-08-10:
Ok, thanks for beeing so fast.
Unfortunately, le link above doesn't point to an existing directory...
I tried to 2 others git dated from today but did not had success after flashing: bleu and red leds dimmly on.
Anyway, I am not stopped so far since I found Serial2 trick.

gfwilliams, 2015-08-10:
Strange - looks like I messed up with git :) This one will have a build in an hour or so: http://www.espruino.com/binaries/git/commits/22398e5f3007f4faae7cca59cce7bdc59ac38acf

I did try a previous build from today and it works - are you sure you chose the correct binary?

asez73, 2015-08-10:
Still the same problem on 

```
>reset();
=undefined
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v80.244 Copyright 2015 G.Williams
>Serial4.setup(115200,{tx:C10,rx:C11});
=undefined
Uncaught InternalError: Unknown serial port device.
 at line 1 col 37
Serial4.setup(115200,{tx:C10,rx:C11});
                                     ^
>
=undefined
> 

```

from the [http://www.espruino.com/binaries/git/commits/22398e5f3007f4faae7cca59cce7bdc59ac38acf/espruino_1v80.244_espruino_1r3_wiznet.bin](http://www.espruino.com/binaries/git/commits/22398e5f3007f4faae7cca59cce7bdc59ac38acf/espruino_1v80.244_espruino_1r3_wiznet.bin)

gfwilliams, 2015-08-10:
Argh, sorry - embarassing. Got it working, then did a quick code tidyup before commit and broke it again :)

http://www.espruino.com/binaries/git/commits/286fed3ea0006fa7dd3f10ddc9a8c65023972a22

will be another hour before that's built though

asez73, 2015-08-10:
Ok :+1: that one runned correctly with 

```
Serial4.setup(115200,{tx:C10,rx:C11});
```
Should we wait that you solve the Serial3 problem to close this one?

gfwilliams, 2015-08-10:
Actually, lets close this. I just found there is a bug for Serial3 already :( https://github.com/espruino/Espruino/issues/409

closed 2015-08-10 by gfwilliams