espruino/Espruino #561: Object doesn't return its value

WebReflection, 2015-08-12:
`Object({any:'value'})` should return the passed argument as is, in Espruino it creates a new empty object each time.

This is breaking badly for all those cases when a piece of code would like to fallback as NOOP

```js
var frozen = (Object.freeze || Object)({a: 123});
// should be the frozen version of {a: 123} or just {a: 123}

var dict = (Object.create || Object)(null);
```
and so on. By spec, `Object(any)` should create a new Object wrapper if `any` is not an  object or a function, so that `Object('asd')` should return `new String('asd')` and same is for numbers and booleans.

`null` and `undefined` will create simply a new object.

Thanks

gfwilliams, 2015-08-13:
Ok, thanks - looks like there wasn't actually a constructor defined before.

I've just added one with the behaviour you suggested. If it's an object/array/function it returns it, otherwise it figures out what the type of variable is, finds a constructor with that name, and calls it.

Having said that, at the moment `new String`/etc **don't actually box their values at the moment**, so that behaviour is of really minimal use.

closed 2015-08-13 by gfwilliams

WebReflection, 2015-08-13:
nobody needs new String really ... specially not new Boolean which as object is always true ;-)

However, `Object` can be safely used as no-op fallback, and it's good if it returns the first argument, like the specs say.

commit b1d1a751 (2015-08-13): Fix Object constructor behaviour (fix #561)
commit 7f09317c (2015-08-13): Fix Object constructor behaviour (fix #561)

Conflicts:

	ChangeLog