espruino/Espruino #562: 2 USB endpoints for HID

gfwilliams, 2015-08-17:
At some point we'll allow the full USB descriptor to be fully changed, which will sort out the HID problems on Windows (which doesn't seem to like enumerating the current composite HID+CDC devices).

However it's [been suggested](https://github.com/espruino/Espruino/issues/221#issuecomment-131428582) that it would be good to be able to support two HID devices (eg. mouse + keyboard), so it would be worth trying to arrange that two endpoints could be made available (maybe with a modification to the `E.sendUSBHID` command

t3rminus, 2017-09-01:
This sounds great! I know HID is pretty new, but I bought the pico to develop some custom HID stuff. Don't care as much about 2 endpoints, but being able to change the descriptor and receive data would be amazing.

gfwilliams, 2017-09-01:
If it's just HID you care about then you can do custom descriptors at least - like is done for the USBKeyboard/etc modules: https://github.com/espruino/EspruinoDocs/blob/master/modules/USBKeyboard.js

But yes, we'd have to do wait for this issue to get solved before you can receive

t3rminus, 2017-09-01:
Yep! Been doing that. I meant it would be nice to be able to change the Product/Manufacturer strings, as well.

bmatusiak, 2022-02-27:
i know this is old..  i see there is  E.sendUSBHID(data)  . now we need E.readUSBHID

gfwilliams, 2022-02-28:
For Bluetooth devices we have an `NRF.HID` event (http://www.espruino.com/Reference#l_NRF_HID) so something similar could be done with USB I guess.

However I'm afraid it's pretty low priority at the moment - which device were you looking at using this with?

bmatusiak, 2022-03-01:
pinless pico usb

gfwilliams, 2022-03-01:
Ok, thanks! Just to say the new WiFi almost certainly won't use STM32 (it'll be Nordic nRF52840). I'm not sure what the situation will be with USB HID support, but it'll definitely support 2 way HID via BLE out of the box