espruino/Espruino #571: SSL/HTTPS support

gfwilliams, 2015-09-16:
Looking at polarssl (now mbed ssl) this doesn't actually look too painful. [mini_client.c](https://tls.mbed.org/api/mini__client_8c_source.html) looks like a good example.

Seems like we can just call `mbedtls_ssl_handshake_step` and handle everything in a nonblocking way (which we'd need, because ESP8266 especially uses a JS driver, which requires the interpreter to stay working behind the scenes).

nkolban, 2015-10-29:
The ESP8266 SDK also supports SSL.  I think what needs to happen is that the interface exposed by JsNetwork needs to be extended to support SSL parameters and then the board drivers can support it if they can.

gfwilliams, 2015-10-29:
I'm not sure that's the right way to handle this - we don't want to implement the exact same SSL for CC3000/ESP8266 AT commands/GSM and Ethernet.

Instead, I think we probably want to add something that works on top of the existing API, so the same code can be used for each. That would then 'just work' on ESP8266, even if it wasn't using the ESP8266 SDK.

ebirger, 2015-10-29:
You can have both -- as long as it's a virtual function in the JsNetwork
interface, it could have a 'generic' implementation and a board specific
one, if it supports it.

gfwilliams, 2015-11-26:
Done now. In 1v82. Should work on anything assuming it's compiled in and there's enough RAM.

closed 2015-11-26 by gfwilliams