espruino/Espruino #573: Non-blocking HTTP requests/setIP()/etc

SpenceKonde, 2015-09-18:
Right now HTTP requests, as well as things like wiznet's setIP() and connect() will block until completed, even though we supply a callback. As far as I can tell, during this time we're waiting for an update from the device - so why can't we do that in the background, and just call the callbacks when it's done? 

It's honestly really annoying how a failed network request will hang Espruino for 30 seconds, throwing everything else out of whack. Basically, if you have a device that periodically updates a webservice  in addition to doing other functions (like a network connected thermostat/switch/whatever with buttons for manual override), a bad network connection, or problems on the server will prevent the device from having any functionality, since it's constantly sitting there waiting on a blocking network request. We make requests by supplying a callback.... to a blocking call. It looks like friendly espruinoey javascript - but it blocks just like on a less civilized platform.

gfwilliams, 2015-09-18:
On ESP8266 you shouldn't have issues - that really is nonblocking because its over serial. There's just the minor gotcha that the connect callback gets called even before a connection has been made.

What happens if you connect to the server using its IP address? 

Unfortunately Wiznet's DNS implementation just blocks, so that's probably causing you the biggest problems (although [rewriting it](https://github.com/espruino/Espruino/blob/master/libs/network/wiznet/DNS/dns.c#L135) looks pretty easy).

To fix it properly:

* `JsNetwork.createsocket` implementations need tweaking so they can return that they're working (0x7FFFFFFF might make sense)
* Http/socket stuff needs to be aware of that, and to then keep calling them until they return a socket number.

Then I think we're sorted - but if you care about it, please get stuck in and fix it yourself. **Nobody has committed any fixes/improvements to core Espruino code since January this year**.

It can be compiled, run, and debugged on Linux so you can get all of that working pretty quickly and easily. It's then just the WizNet-specific code you'll have to go onto the board for.

Note: I doubt CC3000 will ever do proper nonblocking because TI's drivers are awful.

SpenceKonde, 2015-09-18:
I connect with the IP address, not DNS names (yet). If the webserver that my desk lamp talks to is down, every request blocks for ~30 seconds, and the device never becomes functional because it's always blocking waiting for requests to timeout. It also seems to hang for a *very* long time if the network cable isn't connected. It doesn't ever seem to give up and report a failure... it just sits there blocking until you plug a network cable in. 

I'm not optimistic about my being able to help fix this in the C code. I can't even figure out where the hell this blocking is taking place; I look at the createsocket function, but I can't see where it could be blocking... I can write Arduino C, but that's barely C; Espruino is a much more complicated project. I have a really hard time navigating the code and figuring out where things are and what the existing calls do. I've never worked on a large project written in C before, or really any large collaboratively developed project before - maybe if I had more experience with that, it'd be easier to pick up. Every time I've looked at trying to modify Espruino's C code to fix something, I just get completely overwhelmed, with no idea where to look or what to do.

gfwilliams, 2022-06-08:
Closing this as it's now async - I believe even for WIZnet (although for WIZnet only, resolving name->IP may be blocking)

closed 2022-06-08 by gfwilliams