espruino/Espruino #577: trying to build on Mac OSX

jcw, 2015-09-26:
Hello - I tried building Espruino on Mac OSX 10.10.5 w/ Xcode 7.0, and get this error:

```
Espruino$ STM32F4DISCOVERY=1 RELEASE=1 make
Generating platform configs
Generating pin info
Generating JS wrappers
WRAPPERSOURCES = src/jswrap_array.c src/jswrap_arraybuffer.c src/jswrap_date.c src/jswrap_error.c src/jswrap_espruino.c src/jswrap_flash.c src/jswrap_functions.c src/jswrap_interactive.c src/jswrap_io.c src/jswrap_json.c src/jswrap_modules.c src/jswrap_number.c src/jswrap_object.c src/jswrap_onewire.c src/jswrap_pin.c src/jswrap_pipe.c src/jswrap_process.c src/jswrap_serial.c src/jswrap_spi_i2c.c src/jswrap_stream.c src/jswrap_string.c src/jswrap_waveform.c libs/math/jswrap_math.c libs/graphics/jswrap_graphics.c libs/network/jswrap_net.c libs/network/http/jswrap_http.c libs/network/js/jswrap_jsnetwork.c libs/network/cc3000/jswrap_cc3000.c
DEFINES = -DGIT_COMMIT=32a66b2a1cc9b2c7043092aa788448dd252ea062 -DNO_ASSERT -DRELEASE -DBUILDNUMBER=" 394" -DUSE_USB_OTG_FS=1 -DUSE_DEBUGGER -DUSE_MATH -DUSE_GRAPHICS -DUSE_NET -DUSE_CC3000 -DSEND_NON_BLOCKING -DUSB -DSTM32F4 -DARM -DLINK_TIME_OPTIMISATION -DFAKE_STDLIB -DSTM32 -DUSE_STDPERIPH_DRIVER=1 -DSTM32F407VGT6 -DSTM32F4DISCOVERY -DSTM32F407xx -DEMBEDDED
CC libs/math/acosh.o
CC libs/math/asin.o
arm-none-eabi-gcc: error: 394": No such file or directory
arm-none-eabi-gcc: error: 394": No such file or directory
arm-none-eabi-gcc: error: 394": No such file or directory
arm-none-eabi-gcc: error: 394": No such file or directory
Traceback (most recent call last):
  File "scripts/build_jswrapper.py", line 163, in <module>
    jsondatas = common.get_jsondata(False)
  File "/Users/jcw/Documents/Playground/Espruino/scripts/common.py", line 128, in get_jsondata
    code = open(jswrap, "r").read()
IOError: [Errno 2] No such file or directory: '394"'
make: *** [libs/math/asin.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [gen/jswrapper.c] Error 1
make: *** [libs/math/acosh.o] Error 1
```

The ARM toochain is 4.9.3 20150529 (release) \[ARM/embedded-4_9-branch revision 224288].

All I can tell, is that the build number somehow gets into the file open error. Editing the Makefile and adding "COMMITS_SINCE_RELEASE=394" after line 95 to get rid of the extra space gets past this error.

The next error I get appears to indicate I need to install some more headers:

```
...
CC src/jsnative.o
In file included from In file included from src/jsutils.c:src/jslex.c14:
:In file included from 14/Users/jcw/Documents/Playground/Espruino/src/jsutils.h:
:17In file included from :
/Users/jcw/Documents/Playground/Espruino/src/jslex.h:17:
In file included from /Users/jcw/Documents/Playground/Espruino/src/jsutils.h:17:
/Users/jcw/Documents/Playground/Espruino/gen/platform_config.h:13:/Users/jcw/Documents/Playground/Espruino/gen/platform_config.h10::13 :10: fatal error: fatal error: 
      
'stm32f4xx.h'       'stm32f4xx.h'file  filenot  notfound found
```

I can't quite figure out how to get past this and would very much appreciate some help.

-jcw

jcw, 2015-09-26:
Ooh, never mind on that second error: I forgot to set the STM32F4DISCOVERY env var again.

gfwilliams, 2015-09-28:
Thanks - that's an interesting one. I wonder if it's actually a difference in `git` then, or I guess it could be `wc`'s behaviour.

Please can you try:

```
COMMITS_SINCE_RELEASE=$(shell git log --oneline $(LATEST_RELEASE)..HEAD | wc -l)
COMMITS_SINCE_RELEASE=`echo $COMMITS_SINCE_RELEASE` # Trim spaces
ifneq ($(COMMITS_SINCE_RELEASE),0)
DEFINES += -DBUILDNUMBER=\"$(COMMITS_SINCE_RELEASE)\"
endif
```

jcw, 2015-09-28:
Thx - I added that space-triming trick, but (to my surprise) the bug seems to just move around:

```
Generating pin info
DEFINES = -DGIT_COMMIT=c3219b20bc933c1e518a762daf633363cfcb7065 -DBUILDNUMBER="OMMITS_SINCE_RELEASE " -DUSE_DEBUGGER -DUSE_FILESYSTEM -DUSE_MATH -DUSE_GRAPHICS -DUSE_NET -DUSB -DLINUX
CC src/jslex.o
```

Here's what I see:

```
Espruino$ git tag | grep RELEASE_ | sort | tail -1
RELEASE_1v72
Espruino$ git log --oneline RELEASE_1v72..HEAD | wc -l
     403
Espruino$ echo `git log --oneline RELEASE_1v72..HEAD | wc -l`
403
Espruino$ 
```

Probably some make expansion weirdness. All-in-one seems to fix it, though:

```
COMMITS_SINCE_RELEASE=$(shell echo `git log --oneline $(LATEST_RELEASE)..HEAD | wc -l`)
```

gfwilliams, 2015-09-28:
Thanks for checking! I just committed the fix.

closed 2015-09-28 by gfwilliams

jcw, 2015-09-28:
Confirmed. Many thanks.

commit 31a506b6 (2015-09-28): Fix Mac build (fix #577)