espruino/Espruino #584: esp8266 build fail

pastukhov, 2015-09-28:
Hi!
my make script is:
```
#! /bin/bash
export ESP8266_BOARD=1
export ESP8266_SDK_ROOT=/home/artem/build/esp-open-sdk/esp_iot_sdk_v1.4.0
#export PATH=$PATH:/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/  it's already in PATH
export ESPTOOL_CK=/home/artem/build/esptool-ck/esptool
export COMPORT=/dev/ttyUSB0
make $*
```
and i'm getting tons of:
```
LD espruino_esp8266_board.elf
src/jsinteractive.o: In function `jsiInputLineCursorMoved':
jsinteractive.c:(.irom0.text+0x6e): undefined reference to `jsvUnLock'
jsinteractive.c:(.irom0.text+0x8f): undefined reference to `jsvUnLock'
src/jsinteractive.o: In function `_jsiInitNamedArray':
jsinteractive.c:(.irom0.text+0xd1): undefined reference to `jsvGetRef'
jsinteractive.c:(.irom0.text+0xdb): undefined reference to `jsvUnLock'
```

gfwilliams, 2015-09-28:
Can you try git pull again? I merged in a pull request this morning that is supposed to fix this

pastukhov, 2015-09-28:
Same error. Here is the full log: http://pastebin.com/HbvdZRtb

pastukhov, 2015-09-28:
My mistake. It is compiling after `make clean` before build.

closed 2015-09-28 by pastukhov

gfwilliams, 2015-09-28:
Great - thanks for the update!