espruino/Espruino #588: Test ESP8266 SPI support

nkolban, 2015-09-28:
The ESP8266 port has not yet been tested for working with SPI.   This work item is for testing and tracking what might be needed to get that working.

nkolban, 2015-09-29:
2015-09-28 - Looking at jshardware.c, I see that there are a bunch of jshSPI functions that have not be mapped/modeled.  Am adding debug statements to them to see if THEY are being driven.

gfwilliams, 2015-09-29:
Yep, that's software SPI. Hardware SPI requires something like:

```
SPI1.setup({sck: ..., mosi: ...})
SPI1.send(...)
```

Where you have SPI1..SPIx, which are hardware peripherals. You define the number in BOARD.py, as well as choosing where they are with the data returned from `get_pins` in there. There's a bit of info on that in `README_BuildProcess.md` I think.

... also, generally I don't advise using the `new Pin(..)` method. Ideally you'd have D0..D31 (or whatever) defined by `get_pins`, and they'd automatically be global variables that you could access just by typing `D12` or similar.

nkolban, 2015-09-29:
2015-09-29 - A big clue has been found in that currently, the `ESP8266_BOARD.py` only defines GPIO pins 0-7.  A new work item has been raised to map this file correctly.  It is suspected that this definition is preventing SPI from working properly.  For example, a request to create `Pin(12)` is failing because Espruino has been told that only pins 0-7 are usable.

closed 2015-09-29 by nkolban