espruino/Espruino #590: Add support for DNS name resolution

nkolban, 2015-09-28:
The ESP8266 port does not yet have working support for DNS name resolution (i.e. taking a hostname and retrieving the IP address associated with it).  This work items is being created to track the resolution of that omission.

nkolban, 2015-09-28:
The ESP8266 has an API called `espconn_gethostbyname()`.  See the ESP8266 SDK for details.  At a logical level, when this API is called, it is **not** blocking.  Instead, it takes a callback function that is invoked when the hostname has been resolved.

When we look at the Espruino board specific interface to DNS resolution, we find the following in `libs/network/esp8266/network_esp8266.c`:

    void net_ESP8266_BOARD_gethostbyname(
       JsNetwork *net,
       char *hostName,
       uint32_t *outIp)

As we see, we have a non obvious mapping between the two capabilities.

gfwilliams, 2015-09-29:
I think you could rely on the fact that `gethostbyname` is always called from the same function as `connect`, just before - so you could actually kick the DNS off from `connect` itself and you'd be fine.

```

dns_name = ""
gethostbyname(name) {
  dns_name = name;
  return -1;
}

connect(address) {
  if (address==-1) dns_then_connect(dns_name)
  else connect(address);
}
```

it's pretty evil though. Ideally there would be a better way of handling this, but it's probably better to get it working for now and to then worry about fixing the API later on.

nkolban, 2015-09-29:
If I can make the assumption that gethostbyname is always linked to connect ... then yes ... I see that will work great ... however, that then removes the current hope to expose a gethostbyname API to the JS user.

HyGy, 2015-10-19:
I'll try it. But with virtual hosts it will not work...

thx,
HyGy

nkolban, 2015-10-20:
Implemented a simple ESP8266WiFi.getHostByName() function.  Documented in the user guide.  Submitted a Pull Request.  Uploaded firmware build that contains the code.

HyGy, 2015-11-03:
What is the state of this in the esp8266 binaries? When will be included?

And when it will incuded how can we use it? 

http.get({
        host: "thingspeak.com",

or stg command, what can resolv the name?

like
ESP8266WiFi.getHostByName('thingspeak.com');

thanks,
HyGy

nkolban, 2015-11-03:
Howdy HyGy ... the latest binaries have a class called "ESP8266".  

You can code:

```
var ESP8266 = require("ESP8266");
ESP8266.getHostByName("thingspeak.com", function(address) {
   http.get({
      host: ESP8266.getAddressAsString(address),
      ...
   ...
});
```
I don't know when (if ever) we will be able to resolve hostnames by string without doing an explicit lookup.

HyGy, 2015-11-03:
Okay, that is good for me like this. I think it is usable, and not difficult.

What if if the same ip address uses multiple host names (virtual hosts for 1 ip address)? How can I add the host name? Somehow in the header?

nkolban, 2015-11-03:
The ESP8266 supplies an SDK that programmers can code against in C.  This is what we are using inside the Espruino port.  There is an API provided that takes a String as input (representing a hostname) and does a DNS lookup and asynchronously calls us back when an IP address is found.  The callback supplied by the SDK returns only a single IP address.  This seems to tell me that a hostname (at least as far as the SDK is concerned) can map to only a single IP address.

HyGy, 2015-11-03:
So!

here is the solution. (maybe I asked wrong)

need to add a header "host", it is works for me my own apache server, i can get all the sites from the same address chainging the Host header in the request header.

```javascript
var http = require("http");
        http.get({
            host: ipAddress,
            port: 80,
            headers: { 'Host' : 'tingspeak.com', },
            path: '/update?key=' + ApiKey + '&field1=' + press + '&field2=' + temp
        }
```

nkolban, 2015-11-03:
Ahhh ... now I see.  Sorry ... I mis-understood the question.   If that ever happens again, please be patient with me and keep asking using different words.  I'll get it eventually :-)

HyGy, 2015-11-03:
All my respect to you, you helpd me a lot, and U make a very great job about porting!!!! I think this can change the iot devices, couse js is easy, its close to real world events.

I used c with arduino before, and other things but I think this is the fastest coding env, with js. And esp8266 is very cheap, so I can connect things together. Like a air quality mesurement, a power flow meter, gas meter, etc!

So you are great! Please keep going!!!

nkolban, 2015-11-11:
There appears to be problems when users specify a string hostname on TCP/IP based functions.  We need to do better than just document not doing this ... even if it is to throw an exception.  See the following thread.

http://forum.espruino.com/conversations/277402/#comment12620455

closed 2015-12-05 by tve