espruino/Espruino #595: Handle ESP8266 reconnectCB events

nkolban, 2015-09-29:
The ESP8266 reconnectCB event which is called when the ESP8266 detects a TCP/IP level error currently does no more than log a message.  This will result in a leak of socket descriptors or stalled code.  This work item asks for the design and implementation of handling the reconnectCB properly.

It is imagined that it will not be too complex ... and that a reconnectCB will "merely" set the state of the socket into its error state which will then result in a clean up in the future.   However since this event can almost literally occur at any time, analysis needs to be done on the state machine to ensure that the state handling is correct.

nkolban, 2015-10-01:
Examining the code, we find that reconnectCB does nothing.  Here are the recommended changes.

First within the reconnectCB, we call `setSocketInError` to flag the socket as now in the error state.  And then we can return from the callback.   Now we can figure out how to handle the error in the Espruino callback architecture.

For a recv ... we will say that if we have no data to return and SOCKET_STATE_IN_ERROR then return -1 which should flag upstream that we can close the socket.

For a send ... if we are in SOCKET_STATE_IN_ERROR then return -1 which should flag upstream that we can close the socket.

For doClose() ... things get a little bit more interesting.  If we have received an async error indication, should we call espcon_disconnect() and if so, will we ever get a callback?  This is research that must be completed before attempting the further design and implementation.  It may turn out that how we handle this will be a function of the error type and not just that an error was caught.

nkolban, 2015-10-07:
A question has been created on the ESP8266 forum to see if we can figure out what might happen to the state of the socket if a reconnectCB happens.

See: http://www.esp8266.com/viewtopic.php?f=9&t=5864

nkolban, 2015-11-12:
While waiting for the above response, we have implemented some basic flagging logic as we are finding that the absence of a handler is more troublesome than getting it right first time.

tve, 2015-11-12:
When you get a reconcb the connection has been closed. It's equivalent to a disconnect callback, except that it tells you that an error has occurred, i.e. it's not a clean remote-party initiated connection close.

nkolban, 2015-11-12:
Thanks @tve  ... so the notion then is that if one gets a reconncb there is no subsequent need to perform an explicit espconn_disconnect()?

tve, 2015-11-12:
Yup. You will get an error if you call disconnect.

nkolban, 2015-11-21:
The understanding that a reconnectCB will **not** be followed by a disconnectCB needs to be handled correctly in the environment.

tve, 2015-12-18:
I believe all this has been fixed with the Wifi and socket libraries revamping.

closed 2015-12-18 by tve