espruino/Espruino #599: Change Doxyfile to include other targets and libs

nkolban, 2015-09-29:
The current Doxygen generation does not include source files that are found in all the targets (boards) nor does it include the libs (such as networking).   This issue looks to change the Doxygen generation to include that information.

nkolban, 2015-10-06:
Changes have been made to the `Doxyfile` controlling Doxygen generation.  The primary changes were:

* No generation of Latex (HTML only now)
* Additional directories included in the Doxygen parsing
* Recursive directory parsing enabled
* Exclusion of markdown files from doxygen parsing

We also have a Wiki page now that contains instructions for Doxygen usage and maintenance for the project.

gfwilliams, 2015-10-06:
Thanks! you know there is also https://github.com/espruino/Espruino/blob/master/doxygen/README.md for Doxygen info?

In this bug it might be worth thinking about:

## Parsing the JSON headers of wrapped functions into Doxygen

Unless Doxygen specifically has an option, the best bet I see is to write a script that reads the C file and parses docs (there's some code in the scripts folder for this), then creates a new H file in a temporary directory with Doxygen-formatted comments in.

That should be enough to trick it into documenting them properly.

## Making `\brief`

There's a flag in Doxygen to use up to the first fullstop as the `brief`. It might be worth having that on if you're finding they're not documented properly right now?

closed 2015-10-06 by nkolban