espruino/Espruino #603: Error while connecting to TCP on second go

samehhady, 2015-09-30:
When uploading this sample sketch everything works fine on the first time, but when i re-upload it again I get this error and the ESP restarts.

```html
ASSERT(pSocketData->state != SOCKET_STATE_UNUSED) FAILED AT libs/network/esp8266/network_esp8266.c:781
  #1[r1,l2] Object {
    #2[r1,l2] Name String [1 blocks] "ÿ"      #3[r1,l2] Object {
        #6[r1,l2] Name String [2 blocks] "timers"          #8[r2,l1] Array(0) [ ]
        #9[r1,l2] Name String [2 blocks] "watches"          #11[r2,l1] Array(0) [ ]
        #14[r1,l2] Name String [1 blocks] "net"          #219[r1,l3] String [2 blocks] "\x04\x00\x00\x00\x00ÿÿÿ"
        #15[r1,l2] Name String [2 blocks] "history"          #19[r1,l1] Array(8) [
            #20[r1,l2] Name Integer 0              #16[r1,l1] String [2 blocks] "echo(0);"
            #25[r1,l2] Name Integer 1              #29[r1,l1] String [9 blocks] "console.log(\"<<\"+\"<<<\"+JSON.stringify(process.env)+\">>>\"+\">>\");"
            #24[r1,l2] Name Integer 2              #39[r1,l1] String [2 blocks] "echo(1);"
            #153[r1,l2] Name Integer 4              #150[r1,l1] String [4 blocks] "var net = require(\"net\");"
            #166[r1,l2] Name Integer 5              #102[r1,l1] String [42 blocks] "function startTCP(){\n  net.connect({host:\"54.68.0.122\", port: 8080}, function(socket) {\n    console.log(\"Connected to TCP\");\n    socket.write(\"GET / HTTP/1.1\\r\\n\");\n    socket.on('data', function(data) {\n      console.log(\"Got: \" + data);\n    });\n    socket.on('close', function(data) {\n    console.log(\"closed\");\n    });\n});\n}"
            #183[r1,l2] Name Integer 6              #180[r1,l1] String [18 blocks] "ESP8266WiFi.connect(\"Sam WIFI\", \"sameh.hady\", function(response){\n console.log(\"connected ;)\"); // Now should be connected\n  startTCP();\n});"
            #157[r1,l2] Name Integer 7              #226[r1,l1] String [3 blocks] "ESP8266WiFi.init();"
          ]
        #146[r1,l2] Name String [2 blocks] "modules"          #145[r1,l1] Object {
            #156[r1,l2] Name String [1 blocks] "net"              #144[r2,l1] ...
 
          }
        #47[r1,l2] Name String [2 blocks] "HttpCC"          #46[r1,l2] Array(1) [
            #273[r1,l3] Name Integer 0              #267[r2,l3] Object {
                #268[r1,l2] Name String [2 blocks] "__proto__"                  #264[r2,l1] ...
 
                #271[r1,l2] Name String [1 blocks] "type"= int 0
                #270[r1,l2] Name String [2 blocks] "#onconnect"                  #240[r1,l1] Function {
                    #232[r1,l2] Name Param "socket"                       undefined
                    #49[r1,l2] Name String [1 blocks] "ÿcod"                      #243[r1,l1] String [31 blocks] "{\n    console.log(\"Connected to TCP\");\n    socket.write(\"GET / HTTP/1.1\\r\\n\");\n    socket.on('data', function(data) {\n      console.log(\"Got: \" + data);\n    });\n    socket.on('close', function(data) {\n    console.log(\"closed\");\n    });\n}"
                    #50[r1,l2] Name String [1 blocks] "ÿsco"                      #227[r3,l1] Function {
                        #229[r1,l2] Name String [2 blocks] "return"                          undefined
                      }
                  }
                #274[r1,l2] Name String [1 blocks] "opt"                  #233[r1,l1] Object {
                    #235[r1,l2] Name String [1 blocks] "host"                      #236[r1,l1] String [2 blocks] "54.68.0.122"
                    #237[r1,l2] Name String [1 blocks] "port"                      #239[r1,l1] Integer 8080
                  }
                #276[r1,l2] Name String [1 blocks] "dSnd"                  undefined
                #278[r1,l2] Name String [1 blocks] "sckt"= int 1
                #290[r1,l2] Name String [2 blocks] "#ondata"                  #220[r1,l1] Function {
                    #223[r1,l2] Name Param "data"                       undefined
                    #285[r1,l2] Name String [1 blocks] "ÿcod"                      #225[r1,l1] String [6 blocks] "{\n      console.log(\"Got: \" + data);\n    }"
                    #289[r1,l2] Name String [1 blocks] "ÿsco"                      #286[r1,l1] Array(2) [
                        #287[r1,l2] Name Integer 0                          #227[r3,l1] ...
 
                        #288[r1,l2] Name Integer 1                          #277[r2,l1] Function {
                            #283[r1,l2] Name Param "socket"                               #267[r2,l4] ...
 
                            #279[r1,l2] Name String [2 blocks] "return"                              undefined
                          }
                      ]
                  }
                #305[r1,l2] Name String [2 blocks] "#onclose"                  #293[r1,l1] Function {
                    #294[r1,l2] Name Param "data"                       undefined
                    #300[r1,l2] Name String [1 blocks] "ÿcod"                      #295[r1,l1] String [5 blocks] "{\n    console.log(\"closed\");\n    }"
                    #304[r1,l2] Name String [1 blocks] "ÿsco"                      #301[r1,l1] Array(2) [
                        #302[r1,l2] Name Integer 0                          #227[r3,l1] ...
 
                        #303[r1,l2] Name Integer 1                          #277[r2,l1]8 2013,rst cause:1, boot mode:(3,6)
load 0x40100000, len 28568, room 16
tail 8
chksum 0x57
load 0x3ffe8000, len 2024, room 0
tail 8
chksum 0xd9
load 0x3ffe87e8, len 25744, room 0
tail 0
chksum 0x33
csum 0x33
rS
```

And the sketch being used is:

```javascript
ESP8266WiFi.init();
var net = require("net");

function startTCP(){
  net.connect({host:"IP", port: 8080}, function(socket) {
    console.log("Connected to TCP");
    socket.write("GET / HTTP/1.1\r\n");
    socket.on('data', function(data) {
      console.log("Got: " + data);
    });
    socket.on('close', function(data) {
    console.log("closed");
    });
});
}

ESP8266WiFi.connect("ap", "pass", function(response){
 console.log("connected to WIFI ;)"); // Now should be connected
  startTCP();
});
```

gfwilliams, 2015-09-30:
@nkolban you probably want to add a `kill` handler [like this](https://github.com/espruino/Espruino/blob/master/libs/network/jswrap_net.c#L42-L46)

The build process will then make sure that the ESP8266 code is called when the device is 'soft-restarted' and then you can clean up state

nkolban, 2015-10-01:
@gfwilliams I studied the code you pointed to.  If I am reading it correctly, this says that when a kill "signal" is received and we have a network, we call the function "socketKill".   This function the calls "_socketCloseAllConnections".  This then calls "_socketCloseAllConnectionsFor()" each of the types of sockets.  That in turn calls "_socketConnectionKill" for each socket and that in turn (finally) calls "net->closesocket()" for each of the sockets.

Since the ESP8266 already implements `closesocket()` to close a socket ... I am not seeing what else needs to be done?

gfwilliams, 2015-10-01:
Actually what I meant wasn't the code, so much as just:

```
/*JSON{
  "type" : "kill",
  "generate" : "jswrap_esp8266_kill"
}*/
void jswrap_esp8266_kill() {
  // this gets called when everything needs shutting down
  // it could tidy ESP8266-related stuff up?
}
```

But now you say that - if all sockets have had `->closesocket` called on them, this is probably just a bug in the esp8266 socket handling code?

In fact my guess is it's shutting down and then **immediately** restarting, so the socket won't have actually properly closed by that point and will still be in a CLOSE_WAIT kind of state? maybe?

nkolban, 2015-10-03:
While studying the problem report ... noticed that the HTTP request isn't right.  Instead of:

```
socket.write("GET / HTTP/1.1\r\n");
```

We should have:

```
socket.write("GET / HTTP/1.1\r\n\r\n");
```

And HTTP request ends with an empty CR/LF combination.

gfwilliams, 2015-10-03:
> socket.write("GET / HTTP/1.1\r\n\r\n");

I'm pretty sure Espruino's HTTP client does this right, or there would be a hell of a lot of problems reported :) `GET` then headers, then `\r\n\r\n` afterwards.

nkolban, 2015-10-05:
This issue might also be related to #619 in as much as #619 talks about what might have to happen for a logical reset which we believe occurs during new script upload/run.

nkolban, 2015-10-07:
After study, I think we have found and resolved the issue.  Here is the story so we can see if this is the correct reason and resolution.

When we wish to perform an `ESP8266WiFi.connect()` call the board code is passed a JsVar which is a callback function to be called when we eventually connect.  The design of the solution was to save a reference to this JsVar as a C global variable.   When the ESP8266 eventually notifies us that we are now connected, we look at the global variable and, if not NULL, invoke the JS level callback.

So far so good.  However ... now we start to get grey.   When the `jswrap_ESP8266WiFi_connect` function is called, it is __believed__ that we need to `lock` the JsVar so that it is not flagged as eligible for garbage collection ... so we do that before saving its reference in the C global variable.  Thinking this through ... we asked ourselves the question "What would happen is a second `jswrap_ESP8266WiFi_connect` call were made?".  If we did nothing, then we would lock a 2nd JS variable and save its reference in the C global ... effectively losing addressability to the first variable without having unlocked it ... resulting in a leak.  To prevent this, we added a guard in `jswrap_ESP8266WiFi_connect` which, when called, checks to see if the global has a value.  If it does, we unlock it so that JS can clean up.

This was where our error got introduced.

Following a reset, all JS variables are no longer present ... however, we still had a reference to a JsVar from a previous `connect` call.  When we called `connect` for a second time, we looked at the global variable, saw it was non NULL and attempted to unlock it ... and bang ... we broke.  We can't unlock a JsVar which no longer exists.

So what then is the correct solution?

Option1: Register a `kill` function which unlocks the JsVar variable and sets the C global to be NULL.
Option2: Register an 'init` which NULLs the C global as when `init` is called, there are NO JsVars.
Option3: Don't use C globals ... but instead look and see if there is another mechanism such as leveraging hidden JsVars.

We need to consult with @gfwilliams.

gfwilliams, 2015-10-08:
Yep, as I'd said in chat - Using `kill` is the way to go.

However you could also look at using events - registering a hander for a certain event with `jswrap_object_on` and then just emitting the event when something happened,

Note that if you don't do it on `kill`, then the JsVar with the extra lock gets saved to flash, and is then unreachable when you init.

nkolban, 2015-10-08:
The current implementation merely does NOT unlock the variable and hence it will leak (this is bad).  Before this issue can be put to bed, the kill logic must be added to explicitly unlock variables at kill time.

nkolban, 2015-10-08:
Logic implementing release of JsVar that was used while processing connects has been added to the `kill` callback.

samehhady, 2015-10-09:
great work @nkolban I confirm it is working now great with the new bins

closed 2015-10-09 by nkolban