espruino/Espruino #604: Merge `targets/esp8266/docs` into existing docs

gfwilliams, 2015-09-30:
Much of it is relevant to general Espruino - for instance:

* `targets/esp8266/docs/65-Espruino-networking.md` should really get rolled into better JSON documentation in the jswrap_ files, such that it works its way into the proper reference pages.
* `targets/esp8266/docs/10-The-Build-System.md` -> `README_BuildProcess.md`

... and so on. I think those files were only temporary anyway, but we ought to track this.

gfwilliams, 2015-09-30:
`targets/esp8266/docs/07-Samples.md` should probably go in `EspruinoDocs` too.

nkolban, 2015-09-30:
Yuppers ... the immediate goal was to get them OUT of the temporary development Github repository and into the master.  This way, the temporary github that was previously created can be destroyed.   To the best of my knowledge, these files were the last assets that were to be harvested.  There shouldn't be any need to bring in anything else.   They are being held in targets/esp8266/doc as there didn't appear to be anywhere better to place them while we worked.

gfwilliams, 2015-09-30:
Nope, that's fine - just good to have an 'issue' for it so they get put somewhere they're more use to people

tve, 2015-12-18:
Most of the docs have been incorporated into the class reference documentation, such as the Wifi class.

closed 2015-12-18 by tve