espruino/Espruino #609: setWatch: add holdTime option

bartmichu, 2015-10-03:
I find the ```setWatch``` function super powerful but there is only one useful thing missing - option to specify time that the pin must be changed until the function gets called.

For example I would use it with magnetic reed switch where there is a different handler for opening the door, and another handler for holding that door open for too long.

I guess it could be done with ```time``` and ```lastTime``` but it would be easier and more intuitive to have it baked in :)

gfwilliams, 2015-10-03:
Have you checked out the `debounce` option? That only fires your handler if the input has been the same for a certain number of milliseconds.

However I'm not sure if you can have multiple watches on the same pin with difference debounce amounts - you could try?

bartmichu, 2015-10-04:
Yes the first thing I tried was ```debounce``` option but for some reason I was only testing the falling edge and it wasn't working like I expected. Tried again with rising edge and it works like a charm, loving ```setWatch``` even more now.

As to multiple watches I tested with this code and didn't find any problems.
```
setWatch(function () {
  digitalWrite(LED1, !digitalRead(LED1));
}, BTN, {repeat: true, edge: "rising", debounce: 2000});

setWatch(function () {
  digitalWrite(LED2, !digitalRead(LED2));
}, BTN, {repeat: true, edge: "rising", debounce: 50});
```

closed 2015-10-04 by bartmichu

gfwilliams, 2015-10-05:
Great, thanks for checking - good to know it works ok!