espruino/Espruino #610: Emulation modifiers for the Espruino build system

nkolban, 2015-10-04:
Since Espurino compiles very cleanly on both Windows and Linux to produce Windows and Linux native executables, this allows us to play with writing Espruino applications on those platforms.   However, when building "new boards" we have been limited to running those builds on the target boards during development.  As an experiment, I create a "shim" for the ESP8266 SDK APIs that were used in the Espruino port and compiled the actual Espruino/ESP8266 build and linked it with the shim.   This worked very well.   However ... what I did was only an experiment in a discardable sandbox.

This work item seeks community and owner input into the proposal to modify the Espruino build system to add a "build for emulation" configuration.

For example ... if I want to build Espruino for deployment on ESP8266, we use "make build=ESP8266" .. but if we added an emulation ability, we could use "make build=ESP8266 emulate_on=windows".   This would compile all the source files for the ESP8266 build but compile them with a windows compiler producing windows object files which would then be linked with the ESP8266 shim to produce a windows executable.

This differs from a potential "make build=windows" or "make build=linux" as we are actually running the logic and flow that is board specific but with a board specific shim.

jpeletier, 2015-10-04:
+1. I did the same for Arduino and it saved literally *tons* of work, simply because I could use breakpoints and debug. The next step to make the deal sweeter was to program an Arduino to be a "slave" of the emulator so that when in the sketch you had things like digitalWrite(), it actually lit up a pin on the slave. It even had SPI... The PC was the brain running sketch code, an actual Arduino was like a puppet following hardware-related orders. Bottom line, I think having this kind of environment is extremely useful.

gfwilliams, 2015-10-05:
Personally I'm against this I'm afraid... Unless there's a pre-existing emulator that the code can be run against - in which case, great (but if it's a proper emulator it shouldn't need the code to be recompiled in order to run).

95% of the bugs I've encountered on STM32 are totally random hardware things, for instance strange peripheral configs, or IRQs interrupting stuff at the wrong times. As such if I'd written an emulator first, that wouldn't have flagged up any issues and would have confused matters.

The linux build is very good at nailing down bugs in the interpreter itself, and I'm all for trying to make code platform independent such that it can be run under Linux where it can be properly debugged. If everything is working right, the hardware-specific shim should itself be very small.

For instance, stuff like:

* Telnet support
* SSL (when it gets added)
* WebSockets

should all be able to be developed, compiled and run under Linux, so can be really easily debugged.

nkolban, 2015-10-05:
I was following on from the concept that was suggested here .. 
http://forum.espruino.com/comments/12540121/

Maybe the phrase "emulator" is confusing us ... rather the phrase "test harness" might be more appropriate.   If we want to run on Linux that executes the ESP8266 jshardware.c algorithms or ESP8266 network code, we could link with this test harness and debug on Linux.  This issue was looking to see if we wanted to allow the build system (Makefile) to accommodate that.

gfwilliams, 2015-10-05:
Ahh, sorry - I don't know. Maybe it's better to keep it as a branch in the main repo? I think there's a very high likelihood that once the ESP8266 network code is working it'll just get left well alone, and the test harness may end up not being used much? :)

As far as non-network hardware, I'm still not quite sure if that'll be worth the effort. Things like the SPI/I2C/etc in jshardware should be extremely small bits of code that just call into the esp8266 APIs, and I think there's not that big a chance of bugs.

nkolban, 2015-10-07:
I have had a go at getting a "test harness" working for the ESP8266.  This provides a C library that is written in C and uses Windows APIs for networking.  I can link this library with a build of Espruino that "thinks" it is being built for the ESP8266_board but which actually compiles to x86 code for Windows execution.  The result is an instance of Espruino that runs on Windows and thinks it is running on ESP8266 by making ESP8266 API calls satisfied by the harness library.

Currently, this code lives in my own private repositories.  My question now is the fate of what should happen to this code.

Option 1: I create a Github repository that supplies the ESP8266 test harness plus a Makefile that can be used to build a clone of the Espruino project.

Option 2: The source files and Makefile for building this Windows running Espruino get included in the Espruino code tree.

Obviously, Option 1 will not impact the Espruino project at all since no new artifacts will be added into the repository.  However it will now mean that folks who find Espruino would have to search further for this effort.

Option 2 is the one that I am thinking we can apply attention upon for this specific Github issue and breaks down into two areas:

Philosophy - do we want this capability as part of the Espruino project
Practicality - if we do want this capability, where within the source tree should it live?

My vote is Option 2 ... with artifacts being found in `targets\<board>\emulator\Windows` and `targets\<board>\emulator\Windows`  with a Makefile and README.md being found in `targets\<board>\emulator\Windows`.   The source code of Espruino as a whole would not know that this capability is present and the master Make system would not be affected.

gfwilliams, 2015-10-07:
Option 2 sounds good - I'd go for `targets/esp8266/emulator`. But do we really need separate folders for Windows and Linux? The rest of Espruino is managing without at the moment.

Personally I'd say only do a Linux emulator - it's a million times easier to get GDB running with a proper Espruino build... I'd say that trying to debug while running under Windows you're just making a rod for your own back, and then making two different builds of the emulator for 2 different platforms... well that's 2 rods, and the Mac owners will make a 3rd rod and hit you with that as well :)

There's always 'Option 3: a branch' but it might not be worth it.

nkolban, 2015-10-07:
Ok ... with your permission I'll create targets/esp8266/emulator in the core project.  As for why Windows separate from Linux ... it is my understanding that Windows networking is different from Linux  ... but I'll study deeper and see if I can't find a way to create something common.

gfwilliams, 2015-10-07:
When I was doing stuff, windows networking was pretty similar - you just had to explicitly initialise it in Windows and you didn't in Linux. But either way it should just be an #ifdef really.

But yes, go ahead and put those changes in.

nkolban, 2015-10-08:
The emulator stub files have been added to the source tree.  Now we need to write a Makefile to build it.  I believe that this issue should be suspended until the resolution of #615 which will have an effect.  For example, if #615 provides a rule to create the wrapped source files from the JSON, then this can be leveraged by the Makefile for building the emulator.  In fact, thinking further, if #615 is modular enough, then this issue might simply be some "parameters" for #615.

MaBecker, 2016-03-01:
@nkolban or @gfwilliams can this be closed ?

closed 2016-03-02 by nkolban