espruino/Espruino #614: E.toString() can screw up strings...

SpenceKonde, 2015-10-05:
Latest pico cutting edge: 

```

>tstr=E.toString([(0>>8)&255,0&255,"asfd".length]);
="\x00\x00\x04"
>tstr
="\x00\x00\x04"
>tstr+="test"
=
Uncaught InternalError: Variable type cannot be converted to string
>tstr
=
Uncaught InternalError: Variable type cannot be converted to string
> 

```

This is nasty - E.toString() is an *incredibly* useful function!

gfwilliams, 2015-10-05:
Do you think that caused the assert fail that you saw in the other bug as well?

Pretty sure I know what this is - I think it might have made a flat string, but then couldn't append to it (which should have been possible)

SpenceKonde, 2015-10-05:
Yes, this is what caused the assert fail on nucleo. But it reproduces everywhere.

gfwilliams, 2015-10-05:
Great - thanks! It'll hopefully be pretty easy to track down then. I'll see what I can do today

commit bd986c66 (2015-10-05): Fix bug when appending to a flat string (fix #614)
closed 2015-10-05 by gfwilliams