espruino/Espruino #615: Makefile tidyup

gfwilliams, 2015-10-05:
The `Makefile` was pretty bad before, but it's just getting insane now we've stuck ESP8266 and NRF5x in there.

I'd really like to tidy this up...

* Separate files for the major targets (STM32/ESP8266/Linux/NRF5x/others?)
* Maybe a file for each 'library' - eg. Graphics/Network/etc
* As little duplication as possible
* More stuff defined by `BOARD.py` - like whether the board itself includes certain libraries (the `USE_x stuff).
* Remove the explicit `USE_X` checks and somehow make it load the makefile of the given name automatically?
* No makefile-making tools (sorry, I don't think they're needed and they just obfuscate the whole process)
* Maybe change to `BOARD=X make`, but I don't really want to have to run `./configure` or anything first/

Any ideas about how to sort this out? I'm thinking:

* Makefiles in `libs/*`
* Makefiles in `targets/*`
* Get rid of all the `DEFINE=$(shell python some_random_fn)` code and maybe try and work out the basic stuff directly from one Python file (not sure if that can be done nicely).

nkolban, 2015-10-05:
Although this is very low priority, I'd like thought to be given to building under Windows and Mac.  On Windows we can pre-req MINGW which is what we can build with today.  However, unless there are compelling reasons, I'd like to suggest that we not make decisions which would absolutely prevent building on Windows.

gfwilliams, 2015-10-05:
Yes, totally.

I think we're going to need Make and Python at a bare minimum, but after that we should try and make it build without many extra dependencies. It will currently build on OpenWRT-based platforms too and I'd like to keep it that way (so it doesn't need cross compilation).

nkolban, 2015-10-07:
I'd suggest that we draw a flowchart or sequence diagram to capture the major "parts" of the build process.  For example "process wrappers" -> "compile common code" -> "compile included modules" -> "compile board specific" -> "link edit all together".   If this were drawn prettily we could use this as a reference.   Then, we could start thinking about inputs and outputs of each of these "parts" plus identify conventions for common configuration needed for each one in the form of environment variables and such.  From there, we might be able to design "Makefile" units that are responsible for each step.

That would be my suggested approach .... before reaching for a coding pad and starting to type :-)

gfwilliams, 2015-10-07:
I'm a bit busy for that at the moment, but I did spend quite a while trying to detail the whole process here a week or so ago: https://github.com/espruino/Espruino/blob/master/README_BuildProcess.md

nkolban, 2015-10-12:
Should a `clean` also clean a previous Doxygen generation?  My vote would be yes.

gfwilliams, 2015-10-12:
Could do, I tend to use `make clean` a lot though - even a tiny header file change can really mess things up otherwise - and maybe I wouldn't want the docs to disappear as well

gfwilliams, 2017-06-08:
Finally cleaned the makefile

closed 2017-06-08 by gfwilliams