espruino/Espruino #617: change esp8266 idle loop

tve, 2015-10-05:
Understand http://forum.espruino.com/conversations/275267/#comment12549263 and make changes accordingly. Probably best wait until SDK 1.4 issue is clarified too.

tve, 2015-10-16:
The current work-around posts a message to a task queue just before returning from the main loop so it gets invoked again immediately. That causes a 268us "gap" in processing. I think in general that's OK because espruino will not return if it has work to do, so this is just during true idle periods. But it would still be good to be able to control this better.

tve, 2015-10-22:
I finally got a reply from Espressif. The solution is (quoting them):
>The priorities of 3 user tasks are 22/23/24, so if use it like your way, it will cause some SDK internal tasks of which priority is lower than 22 can not be executed.
>
>Please try :
>1. change
>`system_os_task(loop, USER_TASK_PRIO_0, loopQueue, 10); //internal prio 22`
>to
>`ets_task(loop, 1, loopQueue, 10); `
>2. change
>`system_os_post(USER_TASK_PRIO_0, 0, 0);`
>to
>`ets_post(1, 0, 0);`

I'll have a go at that and measure the effect one of these days...

MaBecker, 2019-02-11:
@tve, can this be closed?

closed 2019-03-11 by MaBecker