espruino/Espruino #618: Handle WiFi connect while already connected ...

nkolban, 2015-10-05:
There appears to be an area of function that needs research and study.  In the ESP8266, we now have a function called `ESP8266WiFi.connect()` that connects the device to a WiFi AP.  This seems to work fine on a virgin ESP8266 boot but seems to fail for scripts that attempt to issue a connect the **next** time the script is run.   A suspicion is that the ESP8266 realizes that it is already connected and the underlying  ESP8266 `wifi_station_connect()` is treated as a no-op.  This also means that the callback functions that are meant to be invoked when the connection to the AP is established don't happen ...  and THAT prevents the callback function registered by the JS programmer when they use `ESP8266WiFi.connect()`  from being called.

We appear to have a couple of choices.   One would be to look and see if we are already connected to the target AP and if so, do nothing but drive the callback that says we are connected.   The other choice might be to perform all the disconnect logic and THEN perform the current connection logic.

I don't know which of these is yet the desired semantics.   Discussion on this is open and comments here are welcome.

jumjum123, 2015-10-05:
After connect I expect to have a clean connection. Therefore I would prefer to disconnect and connect again.

igrr, 2015-10-05:
When user calls `ESP8266WiFi.connect()`, you might not be connected to the AP yet (connection might be in progress). Still, you can get current config from SDK by calling `wifi_station_get_config`. If both SSID and PSK match to those provided by user, do nothing and wait for WiFi callback to fire (there is an issue with WiFi status callbacks in SDK 1.4 reported by @tve; you could poll WiFi status until the issue is resolved).
If SSID or PSK do not match, call `wifi_station_set_config` as usual.

tve, 2015-10-05:
@igrr: +1
@jumjum123 can you please define "clean connection"? Or "non-clean connection"? And if you want to  see a disconnect and reconnect it's asking too much for you to call disconnect and then connect?

To me it's absolutely essential to let the SDK do its persistent connection management. It needs to be able to maintain connections through sleep periods and such where a simplistic model of calling "connect" is insufficient. In many cases the SDK has a connection re-established long before any JS code runs and without doing a scan like a "connect" from a disconnected state requires. Those are critical features when operating on battery.

jumjum123, 2015-10-05:
Clean connection in my understanding is:
- nothing left from previous connect
- no open sockets anymore
- nothing (accidently ?) still in use
Sometimes during a process I may switch from one network to another one. I don't oversee all possible problems which may happen in case like this. From my understanding disconnecting and connecting again seems to be more stable, than reuse an "old" connection.

tve, 2015-10-05:
Seems to me that if you want to disconnect & reconnect then you should write that explicitly. Unless you explicitly disconnect there is no reason that sockets should be reset. The fact that you loose an association doesn't mean that sockets need to be reset, you might reassociate the very next millisecond and as long as the DHCP lease has not timed out the device does own that IP address.

If you think ahead to having a wirelessly connected IDE then you definitely want the persistent connectivity and you wouldn't want your IDE to be disconnected all the time...

igrr, 2015-10-05:
FWIW, whenever the module is disconnected from WiFi network, all sockets (or rather *pcb*s, in LwIP speak) are disconnected because the entire interface is taken down via a call to `eagle_lwip_if_free`.

tve, 2015-10-05:
Yup, I thought I saw that happening. Sounds like another inefficiency in the SDK 'cause I don't think it's really required. I wonder how it deals with roaming from one AP to another (of the same SSID)... I hope other platforms are better at this...

nkolban, 2015-10-06:
As a first step ... some debug has been added to `jswrap_ESP8266WiFi_connect` to log the connection status to UART1 before a new connect request is made.  Lets see what we see ....

closed 2015-10-06 by nkolban