espruino/Espruino #619: Implement ESP8266 specific reset in jshardware.c

nkolban, 2015-10-05:
In a recent update, the `jshReset()` function was changed to perform a hardware reset of the ESP8266.  This is believed to not be a good idea.   This issue asks for study of the `jshReset` intent and implement it to achieve its semantic goals.

In the mean-time, the implementation is being changed to **not** perform a hardware reset.

nkolban, 2015-10-06:
Studying the code and docs, it seems that `jshReset` is supposed to reset peripherals back to their power-on state.  For an ESP8266, it isn't clear what that might mean?  Perhaps we should set all GPIOs to input?  Maybe close all open socket connections?  Maybe disconnect the station?  Maybe stop being an access point?  Maybe disconnect any access point connected clients?

gfwilliams, 2015-10-06:
Definitely put all GPIOs back to their power-on state. Socket connections that originated from JS should already get killed by Espruino.
 
There was some talk of making `connect` take an argument that stated whether it was persistent or not, and if it's not persistent I think we should knock WiFi down too.

gfwilliams, 2015-10-06:
Having said that, for now I think having jshReset reset the GPIO pins (and any hw peripherals) is more than good enough

nkolban, 2015-10-06:
As a workaround for just now ... before pushing a 2nd script from the Web IDE, issue an ESP8266 restart by rebooting the ESP8266.

gfwilliams, 2015-10-06:
It's pretty harsh, but you could:

```
function reset() {
  while(1);
}
```

As I understand it, right now that'll cause the watchdog timer to reset after a second and will totally reboot the board.

nkolban, 2015-10-06:
Unfortunately, I don't think that will work.  The reset is called when a new push of a JavaScript is called from the Web IDE ... what that means is that there is a new script "in flow" over UART.  When the reboot completes DURING the flow of the script, Espruino sees the tail end of the script and tries to interpret it from there ... and we are all broken.   ESP8266 has a "restart" API ... so we can force a restart of ESP8266 from software.

nkolban, 2015-10-09:
Although `jshReset()` has not been implemented in the ESP8266 (i.e. the function is present but returns without doing anything special) ... we think that this will be acceptable for just now.  With the removal of the ESP8266 restart API call, we are now able to push a 2nd (and 3rd ...) script without issue.  As such, we no longer have the severity of bug, as function is fully present without obvious issue.  However, this issue should remain open until we explicitly determine that nothing needs to be done to support a logical reset in this function.

nkolban, 2015-10-11:
User @nog3 has been testing with GPIOs and seems to have found that the pins are in indeterminate states on power on.  This could be because the pins are multiplexed ... they serve multiple duties beyond being GPIO ... but it could also be because we haven't done a reset of the pin states to input when a board reset is requested.

tve, 2015-10-13:
>  the pins are in indeterminate states on power on
That is not correct as far as I know. The Espressif document `0D-ESP8266__Pin_List__Release_15-11-2014.xlsx` describes the power-on state. For most GPIO pins this is off with a weak pull-up. We definitely need to implement jshReset() to put all pins back into OFF (input) state with weak pull-up and ensure all interrupts and timers are off.

tve, 2015-12-05:
jshReset() has been fixed in #750

closed 2015-12-05 by tve