espruino/Espruino #624: Add SNTP support

nkolban, 2015-10-07:
SNTP is the Simple Network Time Protocol.  It allows a network connected device to find the current real time over the internet.  The ESP8266 has SNTP built in.  This means that the ESP8266 can determine the current real time.  However, it doesn't yet appear that this ability of the ESP8266 has been taken advantage off.  This issue seeks to examine how we might leverage SNTP capabilities and whether those should be ESP8266 specific or generalized for Espruino as a whole.

First let us look at what the ESP8266 offers.  It has a set of APIs that start with sntp_*().  To use sntp, we tell the environment about 1 or 2 Internet accessible time servers.  These can be supplied either by IP address or hostname.  Next we initialize SNTP.  Finally we can ask sntp for the current real world time in seconds since the epoch.

Obviously, we can't assume that an ESP8266 will always be Internet connected and hence we can't assume that  we can always ask it for the current real-world time.

For an ESP8266 specific implementation, some functions we might have might include:

* `ESP8266WiFi.setTimeServer(<server ip or hostname> [, <server ip or hostname>])`
* `ESP8266WiFi.getCurrentTime()`

If we go this route, would we also want to integrate this into `jshGetSystemTime()`?

What about a non ESP8266 specific environment?  Should the network interface be expanded to  include SNTP support?  Should we architect a library that would provide SNTP support of which ESP8266 might be an implementor and, if so, what should the interface look like?

chevdor, 2017-05-26:
It would also help/apply for #1172

opichals, 2017-06-09:
This is UDP based protocol. Once UDP is implemented in the Espruino's network stack it should be pretty easy to create a small JS module for it.

gfwilliams, 2017-07-19:
When the UDP goes in, IMO this should definitely be a JS library. It'll give a lot more flexibility in terms of when it gets called as well.

opichals, 2017-10-27:
Got a stripped down version of https://github.com/hueniverse/sntp working using Espruino master. Available at https://github.com/opichals/EspruinoDocs/tree/sntp.

Usage as in the original
```js
var sntp = require('sntp');
var options = {
  host: 'pool.ntp.org',
  port: 123,
  timeout: 5000
};
sntp.time(options, function (err, time) {
  if (err) {
    console.log('Failed: ', err);
    return;
  }
  console.log('Local clock is off by: ' + time.t + ' milliseconds');
});
```

The code only works with module minification turned off (most probably due to the used arrow functions).

I am thinking about doing some more stripping before submitting a pull request.

MaBecker, 2018-01-15:
Well SNTP is implemented, documented  and working  on ESP8266

````
sntp_srv="0.europe.pool.ntp.org"; 
require("Wifi").setSNTP(sntp_srv,2);
console.log(Date().toString());
````

So lets remove ESP8266 tag.

@opichals  what do you think about this [ntp-client](https://github.com/moonpyk/node-ntp-client/blob/master/lib/ntp-client.js)

opichals, 2018-01-15:
@MaBecker It has been some time I played with this... The ntp-client you are referring to seems to be doing a real bare minimum (which still can be stripped a little bit) which can definitely be helpful. But once this SNTP JS module is loaded there remains very little RAM to do anything else unfortunately.

wilberforce, 2018-06-07:
@opichals 

https://github.com/opichals/EspruinoDocs/blob/sntp/modules/sntp.js

On the forums there ha s been talk about lack off sntp support on the esp32 - any reason why the module above can't be pulled into master so that it can used on any board with internet?

opichals, 2018-06-08:
The main reason is that the module hasn't been tested properly.

Also it would be worth to strip it down a bit more as there is definitely room for that. Or perhaps even throwing that out and rather porting the @MaBecker suggested [ntp-client](https://github.com/moonpyk/node-ntp-client/blob/master/lib/ntp-client.js) might be a faster way...

PRs welcome :)

closed 2018-12-18 by gfwilliams