espruino/Espruino #626: Generate a memory map of executables when building

nkolban, 2015-10-07:
When building Espruino (for at least the ESP8266), we want to save a corresponding memory map identifying which code is at which area of memory.  When an internal exception occurs, we could then use this map in conjunction with the address at where the exception happened to get a good approximation of where in the source code the exception was thrown.  This could significantly aid problem resolution.

It isn't clear whether this memory map text file should always be generated or only optionally generated.  If optional, what might the Makefile environment flag to be used be called?

gfwilliams, 2015-10-07:
It's auto-generated and is a `.lst`file - of course with the ESP8266 changes it could be that bit got ripped out of the makefile?

nkolban, 2015-10-07:
That's great input.   I'll check to see what the other builds do to create a ".lst" and see if that is still present for the ESP8266 compilation and, if not, add it back in.

nkolban, 2015-10-07:
@gfwilliams was 100% correct.  The ESP8266 portion of the Makefile did NOT contain the rule to build a `.lst` file.  This was added.

closed 2015-10-08 by nkolban