espruino/Espruino #627: Add 'Weak' functions in jshardware.c

gfwilliams, 2015-10-07:
Maybe there are certain functions in jshardware.c we don't care about on some platforms.

For instance `jshGetPinAddress` that I just added. 

It'd be nice to have a `jshardware_weak.c` with weak functions in it that take the place of non-vital `jshardware.c` functions. This would ease porting work significantly - or could just be a fun source of bugs.

https://en.wikipedia.org/wiki/Weak_symbol

nkolban, 2015-10-07:
My gut reaction is "Oh no!!!".   In Object Oriented languages, we have function inheritance and over-rides but those are scope by object type and we can sensibly visualize these in our minds.   This scheme seems to say that there are function over-rides (default implementations) but at a single global scope.

For example, if you use a weak function called foo:

    void foo(); // flagged as weak.

and then in another source file have:

    void foo(); // Not flagged as weak

and then link them all together, the "strong" version is used .... but if you don't link with the strong object file, the weak version is used by default?

My vote would be to not use "weak" vs "strong" capability of the compilers but instead have each board be required to implement the necessary function.

gfwilliams, 2015-10-08:
Anyone else?

My real reason is that basically nobody is going to test what they put in now over all platforms, because there are now so many... So we don;t want to really get it the state where the header files are full of `ifdef`s

nkolban, 2015-10-08:
I'm not sure I understand.
Lets take an example.  I am imagining that a new or changed ability is added to Espruino  for example "Enable that wazzit".   Since a `wazzit` is board specific, that means there has to be separate implementation for `jshEnableWazzit` for each board (in all likelihood).  So ... in the core Espruino code there becomes a call to `jshEnableWazzit()`.  Now ... that would then BREAK all the builds, because at this time, none of them implement `jshEnableWazzit()`.   That is not good.  So, before this can be committed, all the boards have to have an implementation of `jshEnableWazzit()` ... and in this stub implementation, it does nothing but return an error or perform a noop.  The stub implementation is duplicated for each board.  Next, new issues are created ... one for each board that request the implementation of a `jshEnableWazzit` specific for the board.  Over time, the stubs will be replaced by useful function.

Now ... if we contrast this with the weak vs strong linking concept, as I understand it this results in a single stub implementation of `jshEnableWazzit` that is flagged as weak that does nothing or a default or returns an error but which is not board specific.   This satisfies the link.   A board developer can then edit their `jshhardware.c` and add their own implementation and that will override the weak link and then that overridden version will be used.

So ... in this story ... we seem to have two stories ... both of which will work ...

It appears that the weak/strong linking is far less "work" to introduce a new work item ... and ... yes it is.  However, what might its down sides be?

* If one looks in the `jshardware.c` for a board that doesn't have an implementation of `jshEnableWazzit` ... there will be nothing ... no trace that it even might need to be present.

* There is no "pressure" to actually implement a board specific implementation.   My opinion is that forced corrections are better than optional corrections as they will get done.

* The notion of "weak" vs "strong" linking technology may not be known to all C programmers.  Dumb ass that I am, I had never heard of it.

* The "technique" of weak vs strong linking may not be present in non-GCC compilation environments.  While this is not an issue today, it isn't a stretch of my imagination to consider a super-duper-rule-them-all board coming out in 2016 or 2017 which happens to have its own C toolchain which doesn't support strong vs weak.

As always ... these are opinions and @gfwilliams is my General ... his desires and commands are law and obeyed ... willingly and happily.

gfwilliams, 2015-11-26:
Closing this - probably not worth the trouble.

closed 2015-11-26 by gfwilliams