espruino/Espruino #632: Get licence into ESP8266 files

gfwilliams, 2015-10-08:
Hi Neil,

Assuming you're happy with the license, please can you add the following header to the ESP8266 files you've made?

The rest of Espruino files have it, and I'm supposed to have it for MPL.

```
/*
 * This file is part of Espruino, a JavaScript interpreter for Microcontrollers
 *
 * Copyright (C) 2015 Neil Kolban
 *
 * This Source Code Form is subject to the terms of the Mozilla Public
 * License, v. 2.0. If a copy of the MPL was not distributed with this
 * file, You can obtain one at http://mozilla.org/MPL/2.0/.
 *
 * ----------------------------------------------------------------------------
 * This file is designed to be parsed during the build process
    ^^^^ include this in jswrap_*.c
 *
 * A description of the file
 * ----------------------------------------------------------------------------
 */
```

nkolban, 2015-10-08:
Absolutely ... I have no need to have my name attached to the source.  Would it be ok with you if I changed my name to yours in the source files?  You should be the owner of all this stuff.

gfwilliams, 2015-10-08:
Wow, thanks - that'd be great. Maybe put your name down as well though?

nkolban, 2015-10-08:
The changes have been made to the source files and will be included in a commit and subsequent pull request.  There were two files that I couldn't change ... those were `targets\esp8266\hw_timer.c` and `targets\esp8266\hw_timer.h`.  They were introduced by @tve and I don't know their ancestry.   Suggest that we ask @tve to comment upon them and then you can decide what if anything needs to be done to change comments or register a dependency/inclusion.

MaBecker, 2018-01-04:
Hi @gfwilliams  as the pull request is merged, I suggest to close this.

gfwilliams, 2018-01-05:
Thanks! good plan :)

closed 2018-01-05 by gfwilliams