espruino/Espruino #634: Unable to build Espruino on Windows ... common.py

nkolban, 2015-10-09:
When running the build system on Windows, we get a build time error.   This is not new and has been present for all my history of builds but I was able to comment out the "code" that was failing and ran with it commented out as a hack.  Obviously, the hack was never committed.  This issue asks for a proper investigation of the situation and examination of what might be done to resolve the issue.  During a build, we fail with the following error:

```
22:01:00 **** Build of configuration Debug for project EspruinoMaster ****
"C:\\MinGW\\bin\\mingw32-make" clean 
Traceback (most recent call last):
  File "scripts/get_board_info.py", line 40, in <module>
    print(eval(sys.argv[2]));
  File "<string>", line 1, in <module>
  File "C:\Users\IBM_ADMIN\git\Espruino\scripts\common.py", line 349, in get_board_binary_name
    return board.info["binary_name"].replace("%v", get_version());
  File "C:\Users\IBM_ADMIN\git\Espruino\scripts\common.py", line 320, in get_version
    commits_since_release = subprocess.check_output('git log --oneline '+latest_release+'..HEAD | wc -l', shell=True).strip()
TypeError: Can't convert 'bytes' object to str implicitly
Makefile:623: *** Unable to work out binary name (PROJ_NAME).  Stop.

22:01:01 Build Finished (took 1s.179ms)
```

nkolban, 2015-10-09:
If I am reading the code properly (and Python is not a language I know), we appear to be calling a function called `get_version()` from the function called `get_board_binary_name()`.   It is within `get_version()` we are failing.

To be specific, within `get_version()` (which can be found starting at line 315 of the current `common.py`) there is a line which reads:

    commits_since_release = subprocess.check_output('git log --oneline '+latest_release+'..HEAD | wc -l', shell=True).strip()

and it appears to be that line which is failing.

gfwilliams, 2015-10-09:
Ok, just tested and it works - committed it.

... obviously Python devs don't get the whole 'write once run anywhere' thing.

nkolban, 2015-10-09:
There is a second instance of this that also needs addressed.  It currently reads:

```
else: return v+"."+commits_since_release
```

and should be changed to be

```
else: return v+"."+commits_since_release.decode("utf-8")
```

gfwilliams, 2015-10-09:
If you look at the diff above, it's because I put the `decode` bit into `commits_since_release`, so rather than duplicating it, it only happens once.

Does it not work now then?

nkolban, 2015-10-09:
Sorry Gordon ... there are now two problems in the code ... here is what we currently have:

    commits_since_release = subprocess.check_output('git log --oneline '+latest_release+'..HEAD | wc -l', shell=True).decode("utf-8").strip()

The problem is with the variable called `latest_release` within the middle of the statement ... THAT is what is failing ... the code should read:

    commits_since_release = subprocess.check_output('git log --oneline '+latest_release.decode("utf-8")+'..HEAD | wc -l', shell=True).strip()

The second problem is the one I posted a few moments ago ... when we try and concatenate a string with bytes, it fails on windows ... so for example `myStringVar + myBytesVar` fails ... but `myStringVar + myBytesVar.decode("utf-8")` works.

gfwilliams, 2015-10-09:
Ahh, I missed that other decode - I'm not sure I get what the second issue is?

Please can you make the mods needed to the Makefile and issue a PR? I'll then test it here, and integrate if all is ok

nkolban, 2015-10-09:
Commit and pull request for this issue now done.

commit b7f30a64 (2015-10-09): make common.py work on windows (fix #634)
closed 2015-10-09 by nkolban