espruino/Espruino #636: Check that the password for an access point meets minimum requirements

nkolban, 2015-10-09:
Apparently there are rules on password sizes when becoming an Access Point.  This work item asks for study on what those rules might be and then handling of a user program calling `beAccessPoint` and not passing in a "good enough" password.

nkolban, 2015-10-09:
We now check that if a password is supplied, it is at least 8 characters long.

nkolban, 2015-10-09:
Code added to the code base and documentation added to the provisional user's guide.

rakeshpai, 2015-10-12:
Verified. Throws an error for passwords shorter than 8 chars.

```
>ESP8266WiFi.beAccessPoint("Thing", "bad")
Uncaught Error: Password must be 8 characters or more in length.
 at line 1 col 41
ESP8266WiFi.beAccessPoint("Thing", "bad")
                                        ^
=undefined
```

closed 2015-10-12 by nkolban