espruino/Espruino #637: .beAccessPoint shouldn't require password

rakeshpai, 2015-10-09:
It should be possible to create an access point without a password. With the little work I've done with esp8266/arduino, I know that this is possible to do.

Right now, a call to .beAccessPoint without a password (or an empty password) fails:
```
>ESP8266WiFi.beAccessPoint("Thing")
Uncaught Error: No password.
 at line 1 col 34
ESP8266WiFi.beAccessPoint("Thing")
                                 ^
=undefined
```

rakeshpai, 2015-10-09:
Can't seem to add labels. (Mods only?) Modified the title to make it clear that this is about the ESP8266.

gfwilliams, 2015-10-09:
Just done :)

nkolban, 2015-10-09:
In jswrap_esp8266 we have a function called `jswrap_ESP8266WiFi_beAccessPoint` which is where being an access point is controlled.   The design has been changed to no longer reject this function if we supplied no password.  Instead, if no password is supplied then we become an "Open" access point with no access restrictions (password).  In addition, we now check that if a pssword is supplied, it is at least 8 characters long.

nkolban, 2015-10-09:
Code added to the code base and documentation updated in the provisional user's guide.

rakeshpai, 2015-10-12:
Thanks @nkolban. However, I'm seeing some strange behaviour. The password isn't a required field anymore. However, it appears that the SSID passed in isn't honoured, at least the first time `.beAccessPoint` is called. I get an SSID that looks like `ESP_FC6182` or something, in my list of networks. The second call to `.beAccessPoint` seems to use the SSID correctly though.

I can seem to replicate this every time after a reset. Let me know if you need any more details.

nkolban, 2015-10-12:
I have retagged as a bug to look at the code again and see if there is anything obvious to be seen.  Following that, we'll test becoming an access point from a number of different perspectives.  Obviously if you can find a repeatable pattern that demonstrates the issue, we'd love to hear that.

nkolban, 2015-10-24:
The `beAccessPoint` has been reworked into the `wifi.createAP()` function which will be available in a couple of days.  We need to re-examine this issue in the light of #589 changes.

nkolban, 2015-11-05:
@rakeshpai Could I impose on you to re-examine and see if this issue is still present in the latest builds.  The whole area was recently re-done.

closed 2015-11-13 by nkolban