espruino/Espruino #638: ESP8266.beAccessPoint needs a corresponding .stopAccessPoint

rakeshpai, 2015-10-09:
In fact, I'd suggest renaming `.beAccessPoint` to `.startAccessPoint`, or even `.startAP`, and having a corresponding `.stopAP`.

`.stopAP` could take a callback as an argument, if the SDK provides a confirmation that the access point has been stopped.

nkolban, 2015-10-09:
@rakeshpai Good stuff ... however, see issue #589.  We have a work item to address the existence and naming of all WiFi related items.  Can I ask you to check out #589 and see if that is perhaps an encompassing issue for this one?  If it is, please go ahead and close this issue and add your thoughts to the issue/Wiki page where we are discussing the "correct" design of the exposed interface.  Your input is valuable, so if you have time, please come and join.

rakeshpai, 2015-10-09:
I just found the issue and the wiki. I've made a note in the wiki about this. Thanks.

closed 2015-10-09 by rakeshpai