espruino/Espruino #642: Add ESP8266 support for mDNS

nkolban, 2015-10-10:
The ESP8266 SDK provides support for mDNS (Multicast Dynamic Name Server).  This issue asks for and tracks development towards mDNS support.

nkolban, 2015-10-10:
Test code was built which introduces a function called `ESP8266WiFi.mdnsInit()` which hard codes a call to the ESP SDK `espconn_mdns_init`.  This will **not** be exposed in the end ... however, it is useful at this early level of testing.

nkolban, 2015-10-13:
The test of `ESP8266WiFi.mdnsInit()` has been included in a Pull Request.  Further design and implementation of remaining functions still required.

tve, 2015-10-16:
I'm commenting out the mDNS support in PR #666 as a temporary move. I'm not sure what to do. mDNS is very helpful, but I find the esp8266 port to be really tight on memory. I think we need to set-up a little benchmark that streams some data and the see whether we can make it survive after adding mDNS... Unfortunately with/without mDNS is a compile-time decision, not a run-time one.

Bluebie, 2015-11-10:
Multicast DNS runs atop UDP. Why does this need to be in the core firmware? Shouldn't this be something people implement in Javascript as needed, or pull in from NPM?

nkolban, 2015-11-10:
The ESP8266 provides an implementation mDNS as part of its built in capabilities.  As such, this work item merely exposes the inherent capability of the ESP8266 itself.  Nothing special was needed to be added into the Espruino implementation other than the ability to "switch it on or off".  NPM (as I understand it) is a Node.JS concept and separate from the concept of JavaScript as a programming language.

Bluebie, 2015-11-10:
Maybe I'm misinterpreting this, but it sounded like the mDNS functionality in the ESP8266 SDK wasn't being compiled in until the functions were referenced, and that was blowing up the size of the firmware? The "1200 bytes of storage" statement is a bit ambiguous. If it's not using any extra resources when the user doesn't initialize the feature then this change makes a lot of sense to me. :)

Regarding NPM: The Espruino IDE includes an experimental feature to import code from NPM repositories in order to satisfy code dependancies, which would allow users to publish modules that provide extra functionality. It should also allow users to make use of pure javascript libraries that don't depend on unimplemented NodeJS APIs, like pure javascript crypto libraries. NPM isn't just used for NodeJS - it also distributes javascript code used in web apps, and there's no reason embedded targets can't represent another platform.

nkolban, 2015-11-10:
My guess is that the object file contained within the archives supplied as part of the ESP8266 SDK that contains the mDNS functions also contains static storage.  This could be the cache buffer used to hold cached/history mDNS lookups.  When mDNS is link-edited with an application, the object file is included in the ELF and that includes the static storage (which is held in RAM and may be the 1200 bytes).  So simply link-editing the mDNS functions with the Espruino build for the ESP8266 causes the increase in RAM utilization.  Since this cost is not acceptable since we are already woefully light on available RAM, the decision was made to comment out the references to the mDNS functions which results in the object files in the libraries not being included in the ELF which results in the reduction of the RAM usage.

I didn't know about the NPM integration with the Espruino IDE ... that sounds super cool!!!  Sadly I've not had nearly enough time to actually play with Espruino as a user yet ...

I'm starting to be concerned about the relative lack of RAM.  My "unit tests" of apps have all been good but I am starting to sense trouble brewing when it comes to real-work.  For example, I sat down to build a sample project where I wanted to use a micro SD card and realized that we would need to enable the file-system componentry ... only to realize that the RAM cost of enabling that would be too much.  I may have to put aside my Espruino play until I get devices with more RAM available ... at which time, for me, either it will be the best thing since sliced bread or made redundant by the new devices being able to run Linux with all the Linux level apps that implies.

opichals, 2017-01-21:
I am using the 2.0.0 SDK and building using the docker image published here http://forum.espruino.com/conversations/293219/. I have been trying to make this work but I fail to see anything in wireshark nor anywhere in mDNS browsers (testing on the wemos D1 mini 4Mb board).

According to this https://github.com/nqd/esp8266-dev/blob/master/examples/mdns/user/user_main.c I tried to add the register_server and enable calls like this https://github.com/espruino/Espruino/compare/master...opichals:mdns-esp8266-enable but it doesn't have any effect in my case.

From the code I would assume this should work... Any ideas?

opichals, 2017-01-21:
Forgot to mention that the stock 1v91 image from https://www.espruino.com/files/espruino_1v91.zip didn't work for me either.

MaBecker, 2017-01-21:
check this  http://forum.espruino.com/comments/13401539/

opichals, 2017-01-21:
Thank you! Got it working.

So the mDNS works on ESP8266. What is the purpose of this ticket being still open?

MaBecker, 2017-01-22:
@opichals Can you please share the coding you used  to verify that  mDNS works ?

opichals, 2017-01-22:
After some testing it only works during the first few hundred seconds.

I can see the PTR record sent from esp8266 (with ttl set to 300) in the WireShark output. It works for a while (I assume it's for the duration of the record TTL, haven't checked that to be 100% sure) and then no further mDNS packets are sent from the esp8266. The hostname.local resolution stops working after that. I can still use the board with numerical IP address after that without any issues.

As for the .js code I am using I can see nothing special about that. The wifi configuration along with the hostname is set manually and saved with wifi.save() before flashing the http server code which upon reset only waits for the wifi to get connected before starting the server.

```js
var wifi;
function onInit() {
  wifi = require('Wifi');

  wifi.on('connected', () => setTimeout(() => {
    wifi.setSNTP("cz.pool.ntp.org", 0);

    require("http").createServer(onRequest).listen(80);
  }, 1000));
}

function onRequest(req, res) {
  console.log('req', req.method, req.url);
  res.writeHead(404, {'Content-Type': 'text/html'});
  res.end();
}
```

opichals, 2017-01-22:
I assume this might be caused by the above advertised lack of RAM which the mdns caches might exhaust to break things. Not sure there.

Generally I would agree with the @Bluebie's idea to rather do mDNS as a JavaScript module and make it extremely rudimentary (only the PTR record response implementation) at first which would hopefully not consume that much memory and than optionally implement service advertising further. This of course assumes having the UDP broadcast functionality built into the core Espruino (or at least specifically in the esp8266 targets for now). Having UDP (broadcast) would be handy for network discovery features.

gfwilliams, 2017-01-23:
Having it as a JS module sounds like a great idea - someone on the forum has already come up with something basic as well.

UDP is something I'd like to add - and potentially it could be done with the existing network abstraction and a flag in the `connect` call relatively easily.

My only worry is trying to separate packets - if we did it with the existing code then the data would come in as a stream, rather than a series of individual 'packets' :(

opichals, 2017-01-23:
I gave UDP a shot with https://github.com/espruino/Espruino/compare/master...opichals:espruino-udp yesterday. It managed to send a data packet (linux_network.c) but needs more work and proper cleanup.

I follow the node.js loosely. So I created 'dgram' and dgram.createSocket({ host:..., port:... }, function(sckt) {}) method so far. Then sckt.write() worked for me when compiling the linux version (under OSX though).

I intend to create a proper PR as soon as I get it working. As I am new here I would welcome code review or any other advice or hints.

opichals, 2017-01-23:
@gfwilliams Could you please add a link to the forum thread?

Will have to figure the streaming/message approach out.

gfwilliams, 2017-01-23:
Ok, thanks! That looks really good - was this just debug stuff?

https://github.com/espruino/Espruino/compare/master...opichals:espruino-udp#diff-b2fc0e56784ee4ab9a271cb19cb21b89R368

All I'd say is maybe instead of `bool isUDP` you could define an enum (maybe NetCreateFlags)? That way, if we need to add some other feature in the future, we could do it without having to change all the calls for each platform.

gfwilliams, 2017-01-23:
Actually, I might be getting confused with the mDNS.

ClearMemory did some experiments with UDP broadcasts using a hacked up ESP8266 AT command library: http://forum.espruino.com/conversations/293517/#comment13342508

And he also did mDNS - but that was using the ESP8266 AT commands:

http://forum.espruino.com/conversations/293972/#comment13253290

So it looks like he didn't actually make his own mDNS library in JS :(

opichals, 2017-01-23:
forums: ah, yes, I saw that. There is the https://github.com/mdns-js/node-mdns-js where we could take hints from.

code: Yes, that was just debug, I still struggle to understand the code completely. Will clean this up later on.

I actually originally had the NetCreateFlags there, but it appeared to me that it would pollute the net interface with structures that to me appeared not belonging there and I didn't feel like creating another flags enum (there is the ST_UDP, then NCF_UDP...). But I might be wrong. So you think NetCreateFlags would be fine?

gfwilliams, 2017-01-23:
I know what you mean - however the net interface and NetCreateFlags is defined in network.h, so I don't think it's too unreasonable.

Having to have ST_UDP too is a bit annoying (but it's because I had intended to have ST_WEBSOCKET at some point) - in that case, WebSockets wouldn't be a connection type so much as a protocol. Unforunately that's not done yet :)

opichals, 2017-01-23:
There I would vote to drop NetCreateFlags and rather pass the SocketType through.

gfwilliams, 2017-01-23:
Sounds like a good plan - although it could get complicated.

For example you can have:

* HTTP
* Sockets
* HTTP + TLS
* Sockets + TLS
* Websockets? Websockets+TLS

We'd have to change ST_* to a bit field, and check all the states with '&'. IMO it's more sensible to use NetCreateFlags for now, and then to merge NetCreateFlags and NetCreateFlags in a separate pull request

opichals, 2017-01-24:
> We'd have to change ST_* to a bit field

It is already a bit field and is accessed as such IIRC, so dropping NetCreateFlags is quite easy: https://github.com/opichals/Espruino/commit/8e063c26eaa73ed5acb5e20bad70a8afa1dc3f30

gfwilliams, 2017-01-24:
Neat - thanks!

gulfaraz, 2017-01-25:
I'm looking for mDNS advertising capabilities similar to the features available here -
 https://github.com/agnat/node_mdns

Based on this thread, I'm not sure what is the extent of mDNS implemented yet nor the method to verify mDNS from the ESP8266.
I could find only [function `startMDNS`](https://github.com/espruino/Espruino/blob/eabffa40326d653d3d4b1091a536ab0c96dec02c/libs/network/esp8266/jswrap_esp8266_network.c#L1511) which doesn't seem to have advertising.

I've got no idea how adverts work (skimmed through the code and found that there is an 'AT' function that needs to be called to make this happen), but what would be the approach to make the ESP8266 broadcast an advert ?

opichals, 2017-02-11:
I have created a separate issue to track the UDP support, https://github.com/espruino/Espruino/issues/1062.

I would welcome feedback on the plan as well as on the code in the branch there (I changed the code from what I had before).

gulfaraz, 2017-02-12:
I've got a working mDNS start/stop function within the wifi module

Here is the diff -> https://github.com/gulfaraz/Espruino/commit/4a35ce86b651d11d95ad3aef45daa4c2b3b09007

I'm not done with it but would like to get this reviewed to keep the changes in check

I'm planning to expose the hostname as a parameter to the start mdns function

Also, should this be in a different module instead of the Wifi module ?

opichals, 2017-02-12:
@gulfaraz I have been working on this for a while now. My approach is to avoid being ESP8266 specific and rather implement generic UDP support first #1062 and then create simple mDNS.js to handle discovery. I have done some preliminary tests on linux target and I was able to get to a successful dig reply.

gulfaraz, 2017-02-13:
@opichals great! I'll use my changes as a temporary solution till then. Let me know if you need to test stuff on the esp8266 or anything that will help get this out asap.. thanks :)

opichals, 2017-08-25:
In the UDP support PR (https://github.com/espruino/Espruino/pull/1198) I propose to disable the esp8266 specific mDNS support.

Does anything rely on the built-in advertised `espruino` mdns service? Can we drop the esp8266 specific C mDNS support in favor of a JavaScript implementation once available?

wilberforce, 2017-08-28:
You can set the hostname and this is the name advertised via mDns and becomes the hostname in the dhcp table in the router.

gfwilliams, 2017-08-29:
Does keeping it for the moment cause any problems with the UDP support? It seems like something that should maybe be built-in (even if that's as a JS module that is inlined?)

wilberforce, 2017-08-29:
> Does keeping it for the moment cause any problems with the UDP support?

It will do as you won't be able to have two services listening on the same port.

gfwilliams, 2017-08-29:
But only when you pull in the JS mDNS - and that could do a quick sanity check to see if the ESP8266 method exists?

opichals, 2017-08-29:
I agree that this would be nice to have built-in. But I am afraid that the esp8266 doesn't simply have enough RAM to support it inside the whole Espruino firmware. Hence the development of this issue.

For the esp8266 I cut the max UDP receive message size to 268 bytes (empirical value) https://github.com/opichals/Espruino/blob/espruino-udp/libs/network/esp8266/network_esp8266.c#L704 which seems to let it survive on my local network mDNS traffic. We might have to adjust the native (esp-sdk provided) implementation to do the same.

It does not conflict with the UDP support itself. It would work if disable the native mDNS support in the JS module initialization by using some internal API (think E.disableNativeMDNS or alike). /cc @wilberforce

wilberforce, 2017-08-29:
I just re-res do the thread - is it still true that 1200 bytes are consumed if the native mDns is used?
If so, removing it and using the simplied js makes sense to me - and it will work on all platforms...

gfwilliams, 2017-08-29:
I've just released 1v94, so I'd be happy to merge the UDP code in now and start looking at getting it working on other boards.

So am I right in thinking that right now ESP8266's mDNS doesn't really break anything, so we can leave it in before the merge? We can then work on getting the JS mDNS working nicely, and when it's in we can probably pull the native mDNS out before the 1v95 release?

opichals, 2017-08-29:
Ok, thanks! I will remove it from the PR.

gfwilliams, 2017-08-29:
Sorry - if it's already in the PR, let's just leave your PR as-is then. If anyone wants it they can use 1v94 :)

opichals, 2017-08-29:
@gfwilliams no problem :)

MaBecker, 2018-01-12:
@opichals  as far as I understand it is implemented in 1v95.

So please let me know if I can close this.

opichals, 2018-01-12:
Well, only UDP support is implemented. I think this ticket should become about an official mDNS JS module which is not ready yet.

MaBecker, 2018-01-12:
Lets close this because there will be on ESP8266 only implementation for  this

There will be a Espruino solution available for all boards :-)

closed 2018-01-12 by MaBecker

wilberforce, 2018-04-12:
@opichals 

Did you get any futher with mDNS?

Shall I start a new issue?

opichals, 2018-04-18:
@wilberforce Well unfortunately I didn't get to it yet.

I just created https://github.com/espruino/Espruino/issues/1376 to track progress.