espruino/Espruino #648: Support filesystem in flash on esp8266

tve, 2015-10-12:
The initial implementation proposal was to use the existing FAT filesystem implementation in Espruino to stores files in the on-board flash chip. The filesystem would use flash beyond the first 1MB (and possibly some 32KB or so in the first 1MB in the case of 1MB flash chips).

tve, 2015-10-12:
[Copied from http://forum.espruino.com/comments/12562494/]

I've spent the past couple of hours looking at the FAT32 filesystem to use the "internal" esp8266 flash. The characteristics of the serial NOR flash chips are: 4KB block size with the ability to write 256 byte pages (but only erase full blocks). The amount of flash available is 1MB to 3MB for 2MB to 4MB flash chips, and ~32KB for 1MB flash chips (and nothing for the 512KB flash chips).

The problem I'm hitting with the FAT32 implementation is that it really assumes that 512 byte sectors are writable and erasable individually, which is not the case here. Setting the FAT sector size to 4096 would make things work, but then that's the smallest file size, which means that a 1MB filesystem could only support about 200 files and a 32KB filesystem would be a no-go. Write will be really slow because each sector write will consist of an erase followed by a write. On the RAM front the whole thing may blow up 'cause buffers will be 4096 bytes and there just isn't room for even 2 buffers.

One alternative might be to implement sector remapping, i.e., what a wear-leveling controller does, but I don't think we really want to do that. Another alternative is to use a filesystem that is designed for NOR flash, such as SPIFFS: github.com/pellepl/spiffs. The same issue will come up to use the 2MB flash on the emw3165, as far as I can tell.

I may just implement the 4096 sector size FAT32 on the esp8266, but I believe the better solution would be to implement spiffs. Thoughts?

gfwilliams, 2015-10-12:
Do you think there's be a nice way of using the same JS API for FAT32 and SPIFFS? IMO it's quite likely that someone will want to add an external SD card at some point, so it may be that we want to build FAT32 in anyway.

... but then I think the sector size might be a #defined, in which case a 4k FAT32 would be unable to read normal SD cards, so is useless anyway?

IMO another option is just to expose that flash memory via `require('Flash)`, and to then build a basic FS in JavaScript:

* it wouldn't be using any RAM unless it was used
* that same FS (and any improvements) could be used by any device with spare (or external) flash
* Some people might want to access sectors directly (IMO very likely if you're trying to do something single-use)
* There could be a choice of filesystem - for instance someone might just want EEPROM emulation, or to implement some kind of key/value database

tve, 2015-10-13:
I need to look into the spiffs implementation. Using the same JS API would definitely be the best avenue.

nkolban, 2015-11-24:
As part of the FreeRTOS implementation of the SDK, it supplies a SPIFFs file system implementation.  That might be a useful component should we decide to migrate to FreeRTOS.

MaBecker, 2017-07-23:
Adding 'FILESYSTEM' and 'FLASHFS' to ESP8266 is using to much heap space.

Check #1116  for more details.

closed 2017-07-23 by MaBecker